rsta2/circle #6: Raspberry Pi 2 support

meesokim, 2015-02-12:
Is there any plan to support lastest Raspbeery Pi 2? It will be worthy if it is possible.

rsta2, 2015-02-12:
I'm working on it. It is already running - including MMU and USB - but there are some things left to do. I think I will put it on GitHub in one week or two.

meesokim, 2015-02-13:
That's good news. Is it possible to use quad-core feature and NEON instruction?

kika123, 2015-02-13:
@PeterLemon already has a NEON sample

rsta2, 2015-02-14:
My current focus is to support those features available in Circle on the Raspberry Pi 1 on Pi 2 too. So there won't be a multi-core support at the moment. According to the sample of @PeterLemon a special NEON initialization is not needed. The basic FPU initialization already included in Circle is sufficient. So NEON instructions should be useable.

rsta2, 2015-02-15:
Raspberry Pi 2 support is commited. In the end it was less work than I thought before.

rsta2, 2015-02-24:
I think this can be closed. Multi-core support is another issue I will open.

closed 2015-02-24 by rsta2