rsta2/circle #8: umsd1-1 Partion not mounting

Tyguy160, 2015-04-22:
Hi,

I forked your repository and have been compiling with the arm-none-eabi- toolchain to put the circle bare metal OS on my Raspberry Pi. I have been trying to experiment with writing to a 1 GB USB (FAT 32) drive plugged directly into the USB port on my Pi. After using make in the ../sample/15_files/ directory,  the code successfully boots on the Pi, but hangs after the "Compile time" line that is printed to the monitor. I added an "else" to the "if" statement in kernel.cpp that checks to see if the partition is found and it is successfully being found. It seems that the partition is not being mounted properly. Do you have any idea why the code does not get past the mounting "if" statement? Should I be modifying the defined name of the partition at the beginning of kernel.cpp?

I've attached a screen shot of the modified code (2 else statements added). The code never reaches the second if statement. Any suggestions would be greatly appreciated!

[image]

rsta2, 2015-04-23:
There is a "side effect" when you use *m_Logger.Write (..., LogPanic, ...)*. Because this is a system panic message the message will be displayed in red and the system will halt there. So with the code in your attached screen the system will never come to the file system mount even if the partition was found. You should use *LogDebug* or *LogNotice* instead of *LogPanic* for success messages.

Please try sample/07-storage which displays the partition table of the attached USB drive. The first line after the header should be the FAT32 partition ("Type" probably 0B or 0C) and the "1stSector" and "Sectors" values should be valid. Can you check this?

Tyguy160, 2015-04-23:
Thanks for the quick reply! I changed the LogPanic to LogNotice and it is successfully mounting the drive. Now it is hanging in the section where the code is supposed to "show the contents of root directory".

I tried sample/07-storage and am receiving the following from the partition table dump:

00:00:02:40 kernel: # Status  Type  1stSector  Sectors
00:00:02:40 kernel: 1 00        0B      2               1953790
00:00:02:40 kernel: 2 00        00      0                0
00:00:02:40 kernel: 3 00        00      0                0
00:00:02:40 kernel: 4 00        00      0                0

What's the best way to verify that this information is correct and it read the right USB flash drive?

rsta2, 2015-04-23:
Thanks for the reply! So it looks like the physical access to the USB drive works OK and the partition table is also all right. Unfortunately it's difficult to solve why it hangs in the directory listing. You could only insert debug messages at several places to find the exact place where it hangs. That's difficult from here. It may be a bug or the FAT32 partition may be corrupted. You could try to run something like CHKDSK (or fsck.vfat on Linux) on it. Not sure if it helps. Currently I have no other idea. :-(

Tyguy160, 2015-04-24:
Thanks for all of your help. I didn't find the cause of the hang, but I'm assuming it's my drive is having issues. I don't necessarily need to print the contents of the root directory for this project I am working on, so I removed that section and everything else worked flawlessly. Thanks for everything!

closed 2015-04-24 by Tyguy160

rsta2, 2015-04-25:
I think I have found the bug. The evaluation of the HIDDEN directory attribute (called FS_ATTRIB_SYSTEM in Circle) in *sample/15-files/kernel.cpp* was wrong. If there was a file with this attribute in the root directory it did hang. I have pushed a fix to GitHub. Thanks for support!