rsta2/circle #10: Sample 18-ntptime freezes after some seconds if a keyboard and mouse are attached

rsta2, 2015-09-24:
When the sample code starts, you can ping the pi. Between five to twenty seconds after the start, all freezes, no keyboard response, no mouse response and no net response. Sometimes, freezes immediately, sometimes freezes after one minute. I've found that increment the timer interrupt from 100 to 200 or 250 ints/sec, mitigates the problem.

Reported by @jsanchezv

rsta2, 2015-09-24:
A short delay in the main loop in CKernel::Run() also mitigates the problem but does not solve it.

rsta2, 2015-09-27:
I think I have found the problem. After doing some traces using the CTracer class I found a very strange system behaviour occurring. There were nested USB interrupts which should not happen and cannot be handled. The problem could not reproduced on the Raspberry Pi 2.

After checking the relevant paths in the source code (not a lot) I was guessing a side effect and changed the *data cache clean and invalidate operation* before/after an USB slave DMA operation from touching the entire data cache to the DMA data buffer range only which was already the case on the Raspberry Pi 2.

Now I cannot reproduce the problem in a ping test running one hour. It occurred after some seconds before the change. So I suppose the problem is fixed now. But such problems are difficult and further testing may be needed.

jsanchezv, 2015-09-27:
Hi,

This afternoon I've tested the patch sending near 10.000 packets from two
sessions on same IP and a third session from another IP. Around two
uninterrupted hours, without packets lost and a keyboard and a mouse
working correctly.

Great work again!. :)

Thanks by your efforts.
José Luis

rsta2, 2015-09-28:
That sounds very good. Thanks for this valuable testing work!

rsta2, 2015-10-07:
This problem should have been fixed.

closed 2015-10-07 by rsta2