rsta2/circle #11: USB Device Implementation?

Elmi77, 2015-12-01:
This is more a question than a bug report: I wonder if you tried to use the board as USB-device (but not as host like in your examples) in order to e.g. let it appear as serial interface on a connected host PC (USB CDC device implementation).

Anyway: well done, this is a lot of useful code!

rsta2, 2015-12-01:
No, I haven't tried this yet. With the Raspberry Pi Zero this becomes more interesting but (currently) I have no plans for it. Maybe I will re-think this when I got my Pi Zero.

Thanks!

Elmi77, 2015-12-02:
Yes, the Pi Zero is what I had in mind for this...

rsta2, 2016-03-08:
I have checked this. I think the effort to implement USB device mode and relating gadgets is too big to do it. In the meantime there is a solution using the Raspbian kernel for USB device mode. This is explained [here](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=129653). So Circle will stay with the USB host mode implementation.

closed 2016-03-08 by rsta2