rsta2/circle #12: Other Languages Support

NiloofarGheibi, 2015-12-21:
Hello
First Of all, thanks for this awesome codes, I would Like to Display Farsi Text On the Screen.
I found a  Font.h file, I converted a farsi Font Like "B nazanin" to the bitmap hex file, and replaced it, but it did not work, Do you have any Idea or solution for this?

Thank You so much
Niloofar

rsta2, 2015-12-22:
Thanks! The file *lib/font.h* contains a "Latin 1" font for the character codes 0x21 to 0xFF (16 bytes (= pixel lines) per character, 1 byte (= 8 pixels) per line). If you want to display characters of your language you can replace the characters from 0x80 to 0xFF in font.h. Character 0x80 is in line 129 of this file. The standard ASCII codes 0x21 to 0x7F should remain unchanged because they are used for log messages in Circle.

NiloofarGheibi, 2015-12-25:
Thanks for your reply. I'll try it o out