rsta2/circle #13: Could I use SPI like bcm2835 library?

meesokim, 2016-02-16:
http://www.airspayce.com/mikem/bcm2835/

BCM2835 C library has good SPI fuction.
But currently Circle does not provide SPI function at this moment.
Is there any plan to support SPI?

rsta2, 2016-02-17:
I did some development for SPI support and tried to use the SPI/BSC slave device of a second Raspberry Pi for testing the SPI master driver. Unfortunately the SPI slave does not work because of some reason. I can see the SPI signals from the master (using some program based on the sample/11-gpioclock) and it looks right but the slave does not work right.

Unfortunately I do not have other SPI slave hardware to test the master. So I have some untested SPI master code and I cannot include it into Circle because I do not know if it is working in any case.

meesokim, 2016-02-18:
I understand. I can test SPI if you provde your implementation or I am going to port SPI library for my SPI device and let you know the result.

Thank you for your help.

rsta2, 2016-02-18:
If you send me an Email I can send you my source code for the SPI master. You find my Email address in the header of most Circle source code files.

meesokim, 2016-02-18:
This is my email.

meeso.kim@gmail.com

Thank you for your support.

Best Regards,

Miso Kim

rsta2, 2016-03-29:
A driver for the SPI0 master has been added. So this can be closed.

closed 2016-03-29 by rsta2