rsta2/circle #16: question: what is the difference between rpi2 and rpi3?

questor, 2016-05-20:
I've found the following code (at https://github.com/rsta2/circle/blob/master/lib/bcmpropertytags.cpp#L94):

	u32 *pEndTag = (u32 *) (pBuffer->Tags + nTagSize);
	*pEndTag = PROPTAG_END;

#if RASPPI != 3
	CleanDataCache ();
	DataSyncBarrier ();
#endif

	u32 nBufferAddress = GPU_MEM_BASE + (u32) pBuffer;
	if (m_MailBox.WriteRead (nBufferAddress) != nBufferAddress)
	{
		return FALSE;
	}
	
#if RASPPI != 3
	InvalidateDataCache ();
	DataSyncBarrier ();
#else
	DataMemBarrier ();
#endif

I understand why you clear the data-cache and what the barriers are for, what I don't understand is why you don't clear the datacache and to the dsb when it's a rpi3 and after the writeread there's also a difference between the models. I think it's because of the different arm-cores, but what is the important point to know about the difference between the models regarding dmb, dsb and cachebehavior?

rsta2, 2016-05-21:
On the RPi 3 the method of cleaning and invalidating the data cache did not work any more which was find out by empiric tests. As a solution a non-cacheable (coherent) memory region is used for the property mailbox calls which is reserved in the page table. Because this memory region is non-cacheable it does not need to be cleaned or invalidated.

questor, 2016-05-21:
thanks for the explanation, debugging this sounds like real fun.

closed 2016-05-21 by questor