rsta2/circle #17: Gamepad support?

vtrucco, 2016-06-26:
Hi, awesome work! Tested Mouse and Keyboard and it works great. 
There is no support for gamepads ?
Thanks for sharing!

rsta2, 2016-06-27:
Thanks! No, unfortunately there is currently no support for gamepads in Circle.

rsta2, 2016-11-27:
I have ported the USB gamepad driver from USPi to Circle. It is currently on the *develop* branch and will be on *master* branch when Circle 27 is ready.

vtrucco, 2016-11-28:
Hey, thanks!

DFL-git, 2016-12-06:
Just tested and doesn't work for me.

usbdev0-1: Device ven45e-28e found
usbdev0-1: Function is not supported
usbdev0-1: Function is not supported
usbdev0-1: Alternate setting 0 ignored
usbdev0-1: Alternate setting 0 ignored
usbdev0-1: Device has no supported function
dwhxi: Cannot initialize root port
kernel: Gamepad not found

Using an old (but working in PC) X360 wired controller with raspy zero

thanks...

rsta2, 2016-12-07:
Unfortunately the X360 controller needs a special driver. It does not support the generic USB HID interface, which is supported by Circle now. Thus this controller cannot be used with Circle at the moment. I will clarify this.

Matheus-Garbelini, 2018-05-02:
Yes, there is currently an open source simple driver for it (https://github.com/felis/USB_Host_Shield_2.0/blob/master/XBOXRECV.cpp)
but it would needed to be ported to circle, and there's so much code.

rsta2, 2018-05-03:
I have added this to my TODO list including the link. But I cannot make promises. As it looks I will not have time for this in the foreseeable future.

rsta2, 2018-12-01:
There are drivers for the PS3, PS4, Xbox 360 Wired, Xbox One and Nintendo Switch Pro gamepads in the new Circle 37. The gamepads must be connected to an USB port to work. Bluetooth is not supported.

jsanchezv, 2018-12-01:
Another method to have BT support for gamepads is using a Magic-NS or 8bitdo device. The Magic-NS is tested with Circle and a bunch of gamepads (WiiU Pro Controller, Wii controller and Switch's joyscon, DS3 and wireless XBOX360 gamepads) and works like a charm.

Next week I'll receive a 8bitdo device and test it.

rsta2, 2019-03-20:
Gamepad support is fairly complete in Circle now. Bluetooth cannot be supported, because of legal reasons. This can be closed.

closed 2019-03-20 by rsta2