rsta2/circle #18: Circle not working?

doppioandante, 2016-07-10:
Hi, I'm trying to use circle with my raspberry 3.
I could compile the 01 sample (arm-none-eabi- as PREFIX, RASPPI=3 in Rules.mk).
I have copied the boot files in the sd, and the kernel.img generated by gcc.
Unfortunately I can't see anything happening (if I understand correctly, the ACT led is the one near the power led, green).
Am I doing something wrong? How can I know?

rsta2, 2016-07-11:
I have tested the sample/01-gpiosample again on a RPi 3 with the most recent firmware from [here](https://github.com/raspberrypi/firmware/tree/master/boot). It is working. The green ACT LED left to the red power LED is flashing 10 times and then the system restarts.

Please take the files *bootcode.bin*, *start.elf* and *fixup.dat* only from the link above and put them on the first FAT partition of a micro-SD card along with the self build *kernel.img* file. Then use this card to boot your RPi 3.

If this does not work please write an Email to the address noted in most of the Circle source code files. I will send you a tested *kernel.img* file then.

doppioandante, 2016-07-11:
Hi, thanks for the response. I did what you said and I still only see the act led fleshing one time when the raspi is started. I've also tried to install raspian right now and it works, so the board looks ok.
Is the "copying" a simple cp to the SD card folder?
If so I will drop you a mail.

rsta2, 2016-07-11:
Yes, it's a simple *cp* to the SD card folder.

rsta2, 2016-07-18:
I didn't got a mail from you so far and I cannot reproduce this issue.

doppioandante, 2016-07-18:
HI, sorry for letting you wait. I've sent the mail.

mickael-bosch, 2016-07-31:
Hi, 
I had the same issue on a RPI2 and it is fixed now. I did not investigate the issue a lot, all I can say is that on an ArchLinux system, using arm-none-eabi 6.1.1 does not work and using arm-linux-gnueabihf-gcc 6.1.1 works.
I only tried with the 24-hwrandom sample.

rsta2, 2016-08-01:
@mickael-bosch Thanks for your comment. It may help some people in this particular case.

I had some Email conversation with @doppioandante about this issue and it seems the reason for his problem is somewhat different. He did sent me a *kernel.img* file built with his own toolchain and this one is running successfully on my own RPi 3 but not on his RPi 3. I'm guessing there could be a problem with the SD card but unfortunately we were not able to solve this problem so far.

karelv, 2016-10-29:
@doppioandante, I had similar issue when I started. Then I install raspbian on my sd-card.
After I delete files from boot partition and copy the files mentioned by rsta2. Then it worked for me.

doppioandante, 2016-12-07:
Changed SD card and it magically works, I don't think I will have the will to understand why the other one didn't work.

closed 2016-12-07 by doppioandante