rsta2/circle #27: 26-cpustress

koparasy, 2016-11-30:
Hello and congratulation for your awesome project. I am testing the 26-cpustress sample on a raspberry Pi 3b. I am setting all the compilation options for this platform and I compile successfully the 26th sample. The single threaded/CPU version is executing correctly. Afterwards i follow the instructions and i define ARM_ALLOW_MULTI_CORE. I re-complie all the libraries and re-execute the sample. I am not sure, but i think that everything is executed on top of CPU 0. Am I missing something? Is there any implementation in which all 4 cores of my PI simultaneously execute a part of the computation (Simillar to the POSIX pthreads)? Thank you.

rsta2, 2016-11-30:
Thanks!

With sample/26-cpustress you should get 4 separate fractal images on the screen, each calculated by a different CPU core. I re-checked this with the head revision from master branch. It is working. So this is different from sample/17-fractal where each core calculates a part of one single fractal image. On the bottom of the screen there is the chart of the SoC temperature, which is drawn by core 0.

Be sure not to make a typo on *ARM_ALLOW_MULTI_CORE* because this would not cause an error message. Further *RASPPI = 3* should be set in Rules.mk (or Config.mk). *RASPPI = 2* should also work.

koparasy, 2016-12-01:
You are right! I had miss-configured the Rules.mk file.

closed 2016-12-01 by koparasy