rsta2/circle #29: VC4 port

puppeh, 2017-01-01:
Hi,

Is there any interest in porting (some of) these bits to the even-more-bare-metal VC4 side? Some things might be pretty useful there, I think, and could probably be made to work (i.e. "blobless") maybe almost just by tweaking base addresses and recompiling.

rsta2, 2019-07-07:
There was no response in 2 1/2 years and with the new Raspberry Pi 4 available, I don't think there is interest in this. So close.

closed 2019-07-07 by rsta2

tcz001, 2020-10-23:
@rsta2 thanks for bringing this repo, it's very helpful to learn from your code
Sorry to add comment on a closed issue, but we recently are working on a bare-metal OS(non-linux) that want to use the videocore for graphic acceleration, while looking into the addon/vc4/interface in circle, it seems like not mentioning how we should read this part of ported userland, and very few reference in the /samples/, do you have any plan to add more sample to use it on pi?

rsta2, 2020-10-24:
There is another project [libgraphics](https://github.com/rsta2/libgraphics), which provides two more samples. But this is all far from complete. I'm currently not working on accelerated graphics support and I do not have plans on that.

hinxx, 2020-10-24:
@tcz001 , what exactly are you looking for?

As far as I can tell, the vc4 port to circle is complete enough to allow applications that use openGL(ES) to be built upon it. I'm working on a port of one of the openGL based UI frameworks and will offer my code to Circle in the coming days.

For graphics it comes down to initializing Dispmanx and EGL, and that is demonstrated in the samples (e.i. https://github.com/rsta2/circle/blob/4394f8adff299781e3386482acba2cf91e784ee4/addon/vc4/interface/sample/hello_triangle/triangle.c#L130). After that it is all about openGL, and that is not really RPi nor Circle specific anymore.
For input handling (mouse/keyboard) in Circle you can take a look at native Circle samples like `10-usbmouse` and `08-usbkeyboard`.
If you are after looking for something that is not openGL based (just Dispmanx?) on RPi using vc4, there is not much floating around. Nevertheless, here are couple of resources that you may find some examples at:
https://jan.newmarch.name/RPi/Dispmanx/
https://github.com/AndrewFromMelbourne/raspidmx
https://github.com/peepo/openGL-RPi-tutorial
https://github.com/raspberrypi/userland/tree/master/host_applications/linux/apps/hello_pi

tcz001, 2020-10-25:
thanks @hinxx @rsta2 , I was looking for a baremetal usage of the ported opengl interfaces in userland( thought it was designed for a linux driver on raspberrypi upstream ), your information is very helpful thanks for providing the links.