rsta2/circle #30: Connecting GPIO pin to interrupt rising edge gives continuous sequence of interrupt handler calls

smuehlst, 2017-01-06:
I'm trying to attach an interrupt handler to the rising edge of a GPIO pin. My current test setup is a push button with a pulldown resistor attached to GPIO pin 26 of a Raspberry Pi Zero. Pushing the button toggles the pin from LOW to HIGH, which I can verify with a multimeter and a logic analyzer.

The problem ist that the interrupt handler is being called all the time. Even if I connect GPIO 26 directly to GND, this behavior doesn't change. The symptom is a continuous stream of "Interrupt rising edge" messages on the console.

The code is derived from the code in softserial.cpp. The following is the relevant portion of a minimal test example (full project available if necessary). Is there a problem with the code, or do I have a hardware problem?

	CKernel::CKernel (void)
	:	m_Screen (m_Options.GetWidth (), m_Options.GetHeight ()),
		m_Timer (&m_Interrupt),
		m_Logger (m_Options.GetLogLevel (), &m_Timer),
		m_GPIOManager (&m_Interrupt)
	{
	}
	
	CKernel::~CKernel (void)
	{
	}
	
	bool CKernel::Initialize (void)
	{
		bool bOK = true;
	
		if (bOK)
		{
			bOK = m_Screen.Initialize ();
		}
	
		if (bOK)
		{
			CDevice *pTarget = m_DeviceNameService.GetDevice (m_Options.GetLogDevice (), FALSE);
			if (pTarget == 0)
			{
				pTarget = &m_Screen;
			}
	
			bOK = m_Logger.Initialize (pTarget);
		}
	
		if (bOK)
		{
			bOK = m_Interrupt.Initialize ();
		}
	
		if (bOK)
		{
			bOK = m_Timer.Initialize ();
		}
	
		if (bOK)
		{
			bOK = m_GPIOManager.Initialize ();
		}
	
		return bOK;
	}
	
	void CKernel::InterruptHandler (void)
	{
		m_Logger.Write("CKernel::InterruptHandler", LogNotice, "Interrupt rising edge");
	}
	
	void CKernel::InterruptStub (void *pParam)
	{
		CKernel *pThis = (CKernel *) pParam;
		assert (pThis != 0);
	
		pThis->InterruptHandler ();
	}
	
	TShutdownMode CKernel::Run (void)
	{
		m_Logger.Write (FromKernel, LogNotice, "Compile time: " __DATE__ " " __TIME__);
	
		CGPIOPin clockPin(26, GPIOModeInput, &m_GPIOManager);
	
		clockPin.ConnectInterrupt (InterruptStub, this);
		clockPin.EnableInterrupt (GPIOInterruptOnRisingEdge);
	
		while (true);
	
		return ShutdownReboot;
	}

rsta2, 2017-01-07:
I will try to reproduce this. Will need some time.

rsta2, 2017-01-07:
Unfortunately I cannot reproduce the problem. I have tried the source code in your initial comment on a Raspberry Pi Zero. I did only change the line

`CGPIOPin clockPin(26, GPIOModeInput, &m_GPIOManager);`

to

`CGPIOPin clockPin(26, GPIOModeInputPullDown, &m_GPIOManager);`

to enable the internal pull down resistor. When I connect GPIO26 to 3.3V I get one or some more "Interrupt rising edge" message(s) (because the connection is bouncing) but this effect is over after some milliseconds. There may be also some bouncing when the connection is removed.

But I cannot reproduce the effect that the rising edge interrupt is repeated for a longer time range. That means for me the interrupt condition is cleared successfully [here](https://github.com/rsta2/circle/blob/master/lib/gpiomanager.cpp#L129) and there should be some (oscillating) hardware influence which triggers the interrupt again and again.

smuehlst, 2017-01-07:
> Unfortunately I cannot reproduce the problem. ...

Thank you for trying to reproduce my problem.

It looks like I might have a hardware problem then. The strange thing is that a logic analyzer attached to the same pin doesn't show any oscillations, so the Raspberry Pi itself might be broken.

I will try your modification with GPIOModeInputPullDown tomorrow to see whether it makes a difference.

rsta2, 2017-01-08:
Yes, that's strange. I also think there must be a hardware problem with that GPIO pin.

smuehlst, 2017-01-08:
The symptom that I described occured in a larger breadboard setup. I built a minimal test circuit with the push button and the external pull-down resistor  from scratch, and now the above test program works as expected. So the symptom must have been created by noise in the breadboard setup.

Please excuse the confusion. I should have done that isolated test before opening issue.

closed 2017-01-08 by smuehlst