rsta2/circle #31: Is it realistic trying to process a 1 MHz signal with a Raspberry Pi Zero and circle?

smuehlst, 2017-01-29:
This is more a question than an issue.

I'm trying to process signals that have a 1 MHz clock. My ultimate goal is to simulate a device that is attached to the bus of a 6502 processor.

At the moment I'm trying to mirror the 1 MHz clock signal from one input GPIO pin to an output GPIO pin with the following code:

kernel.h:

	#ifndef _kernel_h
	#define _kernel_h
	
	#include <circle/memory.h>
	#include <circle/actled.h>
	#include <circle/koptions.h>
	#include <circle/devicenameservice.h>
	#include <circle/screen.h>
	#include <circle/serial.h>
	#include <circle/exceptionhandler.h>
	#include <circle/interrupt.h>
	#include <circle/timer.h>
	#include <circle/logger.h>
	#include <circle/types.h>
	#include <circle/gpiomanager.h>
	
	enum TShutdownMode
	{
		ShutdownNone,
		ShutdownHalt,
		ShutdownReboot
	};
	
	class CKernel
	{
	public:
		CKernel (void);
		~CKernel (void);
	
		bool Initialize (void);
	
		TShutdownMode Run (void);
	
	private:
		// do not change this order
		CMemorySystem		m_Memory;
		CKernelOptions		m_Options;
		CDeviceNameService	m_DeviceNameService;
		CScreenDevice		m_Screen;
		CExceptionHandler	m_ExceptionHandler;
		CInterruptSystem	m_Interrupt;
		CTimer				m_Timer;
		CLogger				m_Logger;
		CGPIOManager		m_GPIOManager;
	
		void InterruptHandler (void);
		static void InterruptStub (void *pParam);
	
		CGPIOPin clockPin;
	
		bool volatile state;
	};
	
	#endif

kernel.cpp:

	#include "kernel.h"
	
	#include <assert.h>
	
	static const char FromKernel[] = "kernel";
	
	CKernel::CKernel (void)
	:	m_Screen (m_Options.GetWidth (), m_Options.GetHeight ()),
		m_Timer (&m_Interrupt),
		m_Logger (m_Options.GetLogLevel (), &m_Timer),
		m_GPIOManager (&m_Interrupt),
		clockPin(26, GPIOModeInput, &m_GPIOManager),
		state(false)
	{
	}
	
	CKernel::~CKernel (void)
	{
	}
	
	bool CKernel::Initialize (void)
	{
		bool bOK = true;
	
		if (bOK)
		{
			bOK = m_Screen.Initialize ();
		}
	
		if (bOK)
		{
			CDevice *pTarget = m_DeviceNameService.GetDevice (m_Options.GetLogDevice (), FALSE);
			if (pTarget == 0)
			{
				pTarget = &m_Screen;
			}
	
			bOK = m_Logger.Initialize (pTarget);
		}
	
		if (bOK)
		{
			bOK = m_Interrupt.Initialize ();
		}
	
		if (bOK)
		{
			bOK = m_Timer.Initialize ();
		}
	
		if (bOK)
		{
			bOK = m_GPIOManager.Initialize ();
		}
	
		return bOK;
	}
	
	void CKernel::InterruptHandler (void)
	{
		state = clockPin.Read();
	}
	
	void CKernel::InterruptStub (void *pParam)
	{
		CKernel *pThis = (CKernel *) pParam;
		assert (pThis != 0);
	
		pThis->InterruptHandler ();
	}
	
	TShutdownMode CKernel::Run (void)
	{
		m_Logger.Write (FromKernel, LogNotice, "Compile time: " __DATE__ " " __TIME__);
		m_Logger.Write (FromKernel, LogNotice, "Connecting interrupts to clock pin");
	
		clockPin.ConnectInterrupt (InterruptStub, this);
		clockPin.EnableInterrupt (GPIOInterruptOnRisingEdge);
		clockPin.EnableInterrupt2 (GPIOInterruptOnFallingEdge);
	
		unsigned const clockMirrorPin = 10;
		m_Logger.Write (FromKernel, LogNotice, "Mirroring clock input to GPIO pin %u", clockMirrorPin);
		CGPIOPin clockOutPin(clockMirrorPin, GPIOModeOutput);
	
		m_Logger.Write (FromKernel, LogNotice, "Looping...");
	
		bool oldState = state;
		while (true) {
			bool const newState = state;
	
			if (newState != oldState) {
				if (newState) {
					clockOutPin.Write(HIGH);
				}
				else {
					clockOutPin.Write(LOW);
				}
				oldState = newState;
			}
		}
	
		return ShutdownReboot;
	}

I use a logic analyzer to compare the input and the output signal, and the output signal does not keep up with the input signal, e.g. it misses arbitrary transitions of the input clock signal.

I also have  version of the code that switches the output pin directly from the interrupt handling  routine, and that bypasses the search for the right interrupt callback in CGPIOManager::InterruptHandler(), because I know that only one specific pin triggers an interrupt, but that is not much better, and the code is too large to post here.

So my question is, is it realistic trying to implement this with circle at all, or should I bury this idea?

Thank you

rsta2, 2017-01-30:
With your program an 1 MHz clock would generate 2 million IRQs per second. Unfortunately this is much too much for this system. I would estimate a maximum interrupt rate of 50000 (or maybe 100000) IRQs per second for Circle, but not more.

Another problem is, that an access to a GPLEV register, which is read by Circle to determine the actual level of a GPIO pin, does take relatively long. This is caused by the hardware. On the RPi Zero you can expect a maximum sample rate of 10 to 15 MHz if you read the GPLEV register in a tight loop and store it contents directly to memory. Have a look at *sample/11-gpioclock/* where this has been done. Also setting/clearing a GPIO output pin does take time.

I think this is unfortunately not good news for your application.

smuehlst, 2017-01-30:
Thank you for this assesment. I see that I was little bit naive about the possible interrupt rate... 
I already had looked at the 11-gpioclock example, but I guess the polling model would not work for me either.

closed 2017-01-30 by smuehlst