rsta2/circle #32: Open questions about CPWMSoundDevice class

jsanchezv, 2017-02-06:
Hi Rene,

I'm working again with Circle and have some questions that I want comment with you.

I'm using the CPWMSoundDevice class and, examining his code, you say that supports 12 bit samples at 44.1 sampling frequency. But in code I can see:

assert (nBitsPerSample == 8 || nBitsPerSample == 16);

so, the accepted sizes are 8 & 16. I would prefer to send 12-bit samples and the driver don't need resample on-the-fly. If I'm not wrong, the sampling freq. is configured with:

    write32 (ARM_PWM_RNG1, PWM_RANGE);
    write32 (ARM_PWM_RNG2, PWM_RANGE);

at the Run method. PWM_RANGE is defined as:

#define PWM_RANGE        (CLOCK_FREQ / CLOCK_DIVIDER / SAMPLE_RATE)

so, the PWM_RANGE for 44.1 kHz is programmed with value 5668 because the result is truncated to lower integer. With this number as divisor, the resulting freq is 44107.26888 kHz. With 5669 could be 44099.48845, near from the intended sampling freq than 44107.

Anyway, would it be possible to change the sampling freq without recompiling the library?

My problem is that I need to generate the samples on-the-fly, and a frequency as 44099.48 is a problem to generate the samples. Trying different numbers, I've found two divisors interesting to me, that generates freqs. of 25 kHz and 31.25 kHz. I don't know when is possible to change the #define and expect the generated freq. would be the desired. I don't understand very well the MASH filters principle, and I depend of a correct freq to synchronize another things.

At another hand, could exist a method to know the PWM state without polling the driver? (using a callback from DMACompletionRoutine when nTransferSize = 0?, i.e.). I'm calling repeatedly to Playback method but that's very sensible to USB transfers (I guess) and it's easy to hear clicks!, pops!, and all sort of dirty sounds. The worst part is with a PI B+, that can't copes with all the work to be done.

It's very important to my app that can be a reliable method to generate the sound samples and, probably, use the PWM as a fine clock.

Best regards

rsta2, 2017-02-07:
Hi Jose Luis,

I know, this class has limitations. I have to think about, what can be done on this. I will contact you again.

rsta2, 2017-02-21:
There is a new class *CPWMSoundDevice2* in the base library on the *develop* branch (will be on *master* when Circle 29 is ready) which provides a more flexible low level access to the PWM sound device. Its use is demonstrated in the new *sample/29-miniorgan*. The new class uses two concatenated DMA control blocks and buffers which has the advantage that an USB keyboard can be used in parallel without having disturbing clicks on the resulting signal.

class *CPWMSoundDevice* is now implemented using *CPWMSoundDevice2* and does some data conversion only.

rsta2, 2017-03-19:
The new PWM sound generation class is on the master branch now. It has been renamed to *CPWMSoundBaseDevice*.

closed 2017-04-24 by rsta2