rsta2/circle #33: TFTP serve not working

barnamaci, 2017-02-23:
hi
 i try:

C:\Users\kly>tftp -i 169.254.0.111 GET kernel.img kernel.img

I see on circle serial console:
00:12:25.68 tftpd: Transfer timed out
00:12:52.97 tftpd: Incoming read request from 169.254.229.229
00:13:37.98 tftpd: Transfer timed out
00:13:37.98 tftpd: Incoming write request from 169.254.229.229
00:13:43.04 tftpd: Transfer timed out
00:13:43.21 tftpd: Incoming write request from 169.254.229.229
00:13:48.22 tftpd: Transfer timed out

rsta2, 2017-02-23:
This problem is probably caused by the Windows firewall. You have to add TFTP.EXE to the list of programs which are allowed to pass the firewall. This can be done in System Control > Windows-Firewall. After doing this the TFTP transfer works well with the Windows TFTP client at my site.

barnamaci, 2017-02-23:
THX 
working perfectly :)

rsta2, 2017-02-24:
Great! :) So this can be closed.

closed 2017-02-24 by rsta2