rsta2/circle #35: USB Registers

szoshi, 2017-04-06:
Hello,

I am dong some baremetal development for the Raspberry Pi and could find as good as no documentation in the public domain about the USB core except for the base address. I see that you have implemented USB support so I have a few questions:

1) Is there a standard register structure for USB devices or is it specific to implementations?
2) Where can I find the usb register descriptions for the BCM2835?
3) I am trying to develop a very basic usb driver which does not have to be generic at all. Let us just say that I want to use a keyboard as a fancy keypad connected over USB. Are there enough Infos available (I mean without having to reverse engineer code myself) in the public domain for me to even attempt this or should I just drop the idea?

Many thanks,
Szoshi

szoshi, 2017-04-06:
Hello again,

on digging a bit further, I think i have found some info in the file dwc_otg_regs.h.

rsta2, 2017-04-07:
I think you will have to read source code to implement your own driver, because the USB host controller of the Raspberry Pi is specific and a public documentation is not available.

closed 2017-04-09 by rsta2