rsta2/circle #36: p_thread semantic scheduler?

Piping, 2017-04-22:
Hi, I studied your implementation of scheduler.
Compare to the usual `fork-and-join` approach, your scheduler is kind of an polling based approach that the scheduler is activated on `yield` and run the next possible task.

Does my statement correctly describe your scheduler's behavior?

One limitation for this event based approach does not use multicores to explore true parallelism?

Piping, 2017-04-22:
I want to implement a kernel thread library that supports mutilcores. 

Currently I have trouble to write plain `thread_create`,`thread_join`,`thread_exit`,`thread_yield` on top of your CKernel Class. 

Could you give me some advices to implement CTask using those semantics?

rsta2, 2017-04-23:
Circle provides a cooperative non-preemptive scheduler. This was first implemented to allow the networking code run virtually in the background. It requires support of the other tasks running on the system. Each task has to give up the CPU by calling CScheduler::Yield() or CScheduler::Sleep() from time to time, so that the other tasks can run. A task can also be blocked from execution and awaken later.

Circle provides the class CMultiCoreSupport which allows to use the secondary CPU cores of the Raspberry Pi 2 and 3 for dedicated tasks. In an environment with both CScheduler and CMultiCoreSupport, CScheduler and its tasks should run on core 0. So one can do other things on core 1, 2 and 3. I believe, the implementer of an (embedded) application knows better which tasks to run on which core, better than the system can know this on its own.

This implementation of a scheduler in Circle is relatively simple. But because one of my design goals for Circle is, to keep it as simple as possible (to be understandable) it perfectly fits into this. A preemptive scheduler makes many things a lot more complicated. One point here is, that the functions must be re-entrant.

Nevertheless a project to implement an preemptive scheduler using Circle could be interesting. But I had not the time to think about, how this can be managed. It would require some work to find this out and this is not an easy task. So good luck, if you want to do this, but this is not mainline Circle any more.

Piping, 2017-04-24:
Sure. Thanks for your reply.

I am currently trying to implement a scheduler class on top of your CMultiCoreSupport class. I think my roadmap is to figure 
- define thread api that works well, especially the context switch part with time handler interrupt
- implement synchronization objects like Locks and Conditional Variables on top of your spinlock and entercritical(). 
- figure out how to do interprocess interrupt with scheduler class

There are a lot of hardware primitive to support the necessary functions in this case, and I am quite new to Arm. And my goal here is to implement an OS that is up to modern standard. So I hope in the future you wont mind if I ask more questions regard to arm assembly.

rsta2, 2017-04-24:
OK. I let this issue open. I did not write large code sections in ARM assembly, only small handlers and stubs. But perhaps I can help.

rsta2, 2017-08-26:
No question in four months. I think this can be closed.

closed 2017-08-26 by rsta2