rsta2/circle #37: Arithmetic operations of global variables

Prathik4, 2017-04-24:
First of all thanks for providing this huge set of examples.

I'm facing problem while doing floating point arithmetic operations on global variables. The error log is given below,
kernel.o: In function `std::sin(float)':
/usr/arm-none-eabi/include/c++/6.3.1/cmath:449: undefined reference to `sinf'
kernel.o: In function `std::tan(float)':
/usr/arm-none-eabi/include/c++/6.3.1/cmath:506: undefined reference to `tanf'
kernel.o: In function `std::sqrt(float)':
/usr/arm-none-eabi/include/c++/6.3.1/cmath:487: undefined reference to `sqrtf'
../../Rules.mk:74: recipe for target 'kernel7.img' failed
make: *** [kernel7.img] Error 1

If I declare the variables in local scope or as static in global scope then the bare metal program compiles properly.
Do you have any idea why this linker issue are persisting?

rsta2, 2017-04-24:
I have not tried to link the C++ or C standard library with Circle applications. What I have used is pure bare metal Circle stuff. Sometimes the order of the libraries specified on the command line of the linker does matter. But I'm not a tool expert.

vanvught, 2017-04-24:
When using functions from <cmath> math.h, then you need to add the linker option -lm

Prathik4, 2017-04-25:
Thanks for the quick response guys.
Floating point local variables or static global variables are working fine.  Only the variables which are not declared as static in global scope causes those error.
@vanvught I tired adding the -lm linker option but it gives another set of error. Error log is,
/usr/arm-none-eabi/lib/libm.a(lib_a-s_floor.o): In function `floor':
s_floor.c:(.text.floor+0x54): undefined reference to `__aeabi_dadd'
s_floor.c:(.text.floor+0x60): undefined reference to `__aeabi_dcmpgt'
s_floor.c:(.text.floor+0xb0): undefined reference to `__aeabi_dadd'
s_floor.c:(.text.floor+0xe0): undefined reference to `__aeabi_dadd'
s_floor.c:(.text.floor+0xec): undefined reference to `__aeabi_dcmpgt'
s_floor.c:(.text.floor+0x11c): undefined reference to `__aeabi_dadd'
s_floor.c:(.text.floor+0x128): undefined reference to `__aeabi_dcmpgt'
Then to remove this error I tired different linker and compiler options but the error persist.

rsta2, 2017-04-25:
As I wrote, I haven't tried to link the C++ or C standard library to Circle applications. But the result you have got now, is the one I would have expected: It does not work. The standard libraries are very complex software and they build on several runtime functions and start-up code which is not provided by Circle.

If you only need some math functions, you could implement them on your own, like I did [here](https://github.com/rsta2/minisynth/blob/master/src/math.cpp) for my virtual analog synthesizer application. But this implementation has side effects on precision and handling of infinite and undefined values. In your own application you may know, how to handle these effects.

A general purpose math library is a very different thing. I had a look into the GNU C Library and this is huge stuff. Porting this to Circle would be a difficult task and should be done by someone who can handle this.

Prathik4, 2017-04-26:
@rsta2 Thanks for pointing me in the right direction. I'll try implementing the basic math functions that I require for my application.
One last thing I wanted to ask was about newlib. I found out that newlib is C standard library implementation intended for using on embedded systems. So will this be useful with circle?

rsta2, 2017-04-26:
newlib could be interesting and I heard from one guy that he wants to port it for running with Circle. So maybe there will be something in the future.

vanvught, 2017-04-26:
I have ported the newlib for baremetal Raspberry Pi -> http://www.raspberrypi-dmx.org/raspberry-pi-dmx512-rdm/development-environment
This is one part of the exercise. The next step is to implement the calls in syscall.c 
For example : caddr_t _sbrk(int incr)
However, Circle has already its own malloc() and free() functions.
When you want to use console functions from the newlib, then you need to implement int _write(int file, char * ptr, int len)
All other functions in syscall.c can just returning 0, 1 or -1. 
Please note that newlib is dependent on the __aeabi functions. So you still need to link with -lgcc
Newlib did work fine for me in my own development environment. But for people with less tool experiences it is not that easy to work with newlib. Hence I have dropped newlib and with great help from @rsta2 I have implemented the needed C functions.

Prathik4, 2017-04-26:
Thanks @rsta2 and @vanvught for all the support and quick replies. All the information you guys provided was really very helpful and it cleared a lot of queries I had. For time being I'm able to implement and run basic mathematical functions that I require for my application. Hopefully I'll get the things running the way I intend to on the raspberry pi.

closed 2017-04-26 by Prathik4

smuehlst, 2017-04-26:
I have made some inital experiments to port newlib to Circle. For a first attempt I wanted to leave the system calls unimplemented, and in a second step I wanted to implement the write() and open() system calls based on the SD card class of Circle. Here are some of my findings:

- As @vanvught pointed out, Circle already has malloc() and free(). There is an option for building newlib with externally supplied memory management routines. To make this work, one would have to implement realloc() and calloc() on the Circle side.

- Parts of the standard C library are already implemented in Circle. In my experimental build this gave errors because of duplicate symbols. It would require modifications in Circle, for example a build time option with #ifdefs that disables Circle's C library functions, to resolve this. Another option would be to disable these functions in newlib, but this looks more complicated and probably less maintainable to me.

- With disabling the C library functions in Circle there would be a circular dependency if certain newlib system calls were implemented based on Circle classes.

So as @rsta2 already pointed out porting a C library to a bare-metal system is non-trivial, and while newlib looks like a much better option than the GNU C library it is still not a straightforward story...