rsta2/circle #39: webserver memory leak

thepyper, 2017-05-24:
Hi!

I've come across some memory leak problem in web server implementation.
I can reproduce problem on example 21-webserver, on raspberry pi2.
I first tried with ARM_ALLOW_MULTI_CORE + REALTIME settings in sysconfig.h,
then with ARM_ALLOW_MULTI_CORE only setting, then with default settings, problem remain the same.
What I do is flooding webserver with following command from my pc:
curl -s "http://192.168.1.118?[1-10000]"
What happens is that always after 457 successful replies raspberry hangs.

I did some rough debugging, found that:

- with mem_info() I see that things get allocated but never freed, it looks like any connection is taking about 1 block of 262144, 2 blocks of 4096, 1 block of 1024;

- with TCP_DEBUG I've seen that connections stay stuck in FIN-WAIT-1 state, as if TCPTimerTimeWait never get called, so state never goes further to CLOSING and tcpconnection never get deleted;

- what is strange is that this happens only if I continuously do "fast" writing on screen (without scrolling): I continuously call "\b[H" sequence to go home with screen cursor, then call mem_info(); this results in seeing very fast update of memory consumption; in this conditions, example 21 hangs with even 1 connection, and connection last seen state is FIN-WAIT-1;

- if I remove the mem_info() thing, but keep the TCP_DEBUG thing, I see from the tcp log that 1 connection correctly terminates (goes to CLOSED state, and then I see "Deleted TCB" message); a connection flood anyway terminates with some not-well defined hang;

That's what I've seen so far, hope it is useful :)
I may try some further debugging next days...

thepyper, 2017-05-25:
I've done some further test, I'm not sure why I'm seeing so many different behaviours, maybe there is more than one thing I'm not understanding :)

I got the "memory leak" solved this way: I added the following line in tcpconnection.cpp at line 1119:
StartTimer (TCPTimerTimeWait, HZ_TIMEWAIT);

So, when connection stays stuck in FIN-WAIT-2 state (don't know why yesterday was seeing FIN-WAIT-1) it get closed by timeout. Don't know why my pc is not correctly terminating the connection, but now this command works:

curl "http://192.168.1.250?[1-10000]"

Raspberry does not hang, memory consumption stays constant.
I'm no TCP/IP expert, but a timeout on FIN-WAIT-2 state looks reasonable to me, and looking there:

https://benohead.com/tcp-about-fin_wait_2-time_wait-and-close_wait/

it seems it's usually implemented.

rsta2, 2017-05-26:
Thanks for doing these tests! I have to analyse it, which will take some time.

rsta2, 2017-05-28:
I can confirm, that there is a problem regarding memory usage with *sample/21-webserver*, if the server is flooded with HTTP requests. Because a TCP connection allocates relatively much memory, memory will be exhausted after some time, if the requests come in too quick and/or if they are not finished as expected by the TCP state machine. This may lead to a system failure.

I agree, there should be a time-out in FIN-WAIT-2 state. But this will not solve the entire problem. Because a connection remains in TIME-WAIT state after closing for a minute, and the TCP control block for that connection is still allocated at this time, memory may still be exhausted by flooding requests.

I think, professional TCP/IP and web-server implementations have mechanisms, which handle such situations carefully. Please note, that the TCP/IP stack in Circle is still experimental and the web-server support is very simple. Implementing a bullet-proof networking stack is a huge and time consuming task.

I have added this problem to my TODO list for upcoming Circle releases. But unfortunately I cannot promise at the moment, that there will be a general solution for it in foreseeable time.

thepyper, 2017-05-29:
I agree that time-out does not solve the whole problem.. to mitigate i also changed the HZ_TIMER value from 60*Hz to 1*Hz. This makes flooding a bit harder, infact to crash system I need to run at least 3 parallel instances of the "curl" command.

Would be an acceptable solution to limit the number of concurrent accepted active connections to a fixed value? maybe defined in sysconfig.h? I'm thinking of a counter incrementing each time a connection get accepted, and decrementing each time a connection is deleted from memory.
If counter becomes higher than fixed threshold, Accept() method would refuse connection and avoid allocating any memory.
That would limit memory usage (on "server" side) to a fixed amount and would be a quite cheap solution in term of code change.. maybe it's a bit rough :)

rsta2, 2017-05-29:
Good idea! OK, I will implement this for the next release.

rsta2, 2017-08-07:
The problem has been fixed on the *develop* branch now. The fix will be on *master* branch when Circle 31 is ready. Thanks for reporting the problem and for suggesting the fix.

rsta2, 2017-08-26:
The changes are on *master* branch now.

closed 2017-08-26 by rsta2