rsta2/circle #41: Socket: Connection closed in the middle of file transfer

weidongguo, 2017-06-14:
I am currently using the socket api by Circle.

On the Raspi side, I opened a socket and created a TCP connection with a server that is running the "SimpleHTTPServer" module in Python.
I was able to make a HTTP GET request to ask for a file from this server, and receive it completely.

This works for files smaller than ~2MB. 

When I try to receive a file that is 4MB, it will pause after receiving 1408 blocks of the file (each block is 1460 bytes). I have a loop that keeps calling receive() and check whether or not its return value is 0 as  the ending condition of the loop.

When I check on the server side, it says that the connection is closed by the remote host. Therefore the connection was closed by the Raspi side.

Do you have any clue why this is happening?

Thanks in advance.

rsta2, 2017-08-03:
Sorry that my answer took that long. Unfortunately I cannot reproduce the problem. With a test program I can transfer a file with over 6 MB (exactly 6549404 bytes of arbitrary data) without problems in less than a second. I started the server with:

```
python -m SimpleHTTPServer 8000
```

The test program is:

```
#include "kernel.h"
#include <circle/net/socket.h>
#include <circle/net/dnsclient.h>
#include <circle/net/ipaddress.h>
#include <circle/net/in.h>
#include <circle/string.h>

// Network configuration
static const u8 IPAddress[]      = {192, 168, 0, 250};
static const u8 NetMask[]        = {255, 255, 255, 0};
static const u8 DefaultGateway[] = {192, 168, 0, 1};
static const u8 DNSServer[]      = {192, 168, 0, 1};

// wget configuration
static const u8   Server[]   = {192, 168, 0, 158};
static const char Document[] = "/test.dat";

static const char FromKernel[] = "kernel";

CKernel::CKernel (void)
:	m_Screen (m_Options.GetWidth (), m_Options.GetHeight ()),
	m_Timer (&m_Interrupt),
	m_Logger (m_Options.GetLogLevel (), &m_Timer),
	m_DWHCI (&m_Interrupt, &m_Timer),
	m_Net (IPAddress, NetMask, DefaultGateway, DNSServer)
{
	m_ActLED.Blink (5);	// show we are alive
}

CKernel::~CKernel (void)
{
}

boolean CKernel::Initialize (void)
{
	boolean bOK = TRUE;

	if (bOK)
	{
		bOK = m_Screen.Initialize ();
	}

	if (bOK)
	{
		bOK = m_Serial.Initialize (115200);
	}

	if (bOK)
	{
		CDevice *pTarget = m_DeviceNameService.GetDevice (m_Options.GetLogDevice (), FALSE);
		if (pTarget == 0)
		{
			pTarget = &m_Screen;
		}

		bOK = m_Logger.Initialize (pTarget);
	}

	if (bOK)
	{
		bOK = m_Interrupt.Initialize ();
	}

	if (bOK)
	{
		bOK = m_Timer.Initialize ();
	}

	if (bOK)
	{
		bOK = m_DWHCI.Initialize ();
	}

	if (bOK)
	{
		bOK = m_Net.Initialize ();
	}

	return bOK;
}

TShutdownMode CKernel::Run (void)
{
	m_Logger.Write (FromKernel, LogNotice, "Compile time: " __DATE__ " " __TIME__);

	WGet ();

	for (unsigned nCount = 0; 1; nCount++)
	{
		m_Scheduler.Yield ();

		m_Screen.Rotor (0, nCount);
	}

	return ShutdownHalt;
}

void CKernel::WGet (void)
{
	CIPAddress ForeignIP (Server);
	CSocket Socket (&m_Net, IPPROTO_TCP);
	if (Socket.Connect (ForeignIP, 8000) < 0)
	{
		m_Logger.Write (FromKernel, LogPanic, "Cannot connect to server");
	}

	CString IPString;
	ForeignIP.Format (&IPString);
	m_Logger.Write (FromKernel, LogNotice, "Outgoing connection to %s", (const char *) IPString);

	CString Request;
	Request.Format ("GET %s HTTP/1.1\r\n"
			"Host: %s\r\n"
			"User-Agent: wget/0.01 (Circle)\r\n"
			"\r\n", Document, Server);
	if (Socket.Send (Request, Request.GetLength (), 0) < 0)
	{
		m_Logger.Write (FromKernel, LogError, "Cannot send");

		return;
	}

	u8 Buffer[FRAME_BUFFER_SIZE];
	int nBytesReceived;

	do
	{
		nBytesReceived = Socket.Receive (Buffer, sizeof Buffer, 0);
		if (nBytesReceived > 0)
		{
			m_Screen.Write ("#", 1);
		}
	}
	while (nBytesReceived > 0);

	m_Screen.Write ("\n", 1);

	m_Logger.Write (FromKernel, LogNotice, "Closing connection with %s", (const char *) IPString);

	// close is done when Socket is destroyed here
}
```

rsta2, 2017-08-26:
There is a new HTTP client class and *sample/31-webclient* in Circle 31. Perhaps this helps to easier implement HTTP requests.

I have found a problem in the TCP code, which may cause losing data at the end of a TCP connection. This has been fixed. I don't think, this problem may have caused this issue here.

rsta2, 2019-09-04:
This old issue cannot be analyzed any more and it is likely that the problem was solved in the meantime. Therefore the issue is closed.

closed 2019-09-04 by rsta2