rsta2/circle #42: Disable Prefetch.

koparasy, 2017-06-14:
Hello and thank you for your project. I am using your library to characterize the power/energy behavior of the raspPI3b. I am trying to disable the prefetching unit of the CPU. According to the arm [manual](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0500e/BABHGCCE.html) I need to set a value to CPUACTLR when processor is idle. I tried to add  code in startup.S which reads CPUACTLR and then writes back another value to the register. However when i read and print the register the resutls are the same. 
```
mrrc    p15, 0, r2, r3, c15 
bic r2, r2, #28672
orr r2, r2, #253755392
mcrr    p15, 0, r2, r3, c15  
``` 
Could u please provide me some guidance on how to do it?

rsta2, 2017-06-14:
Unfortunately I have no experience in using this register. From the manual I think it should be accessible from the code. Perhaps it is interesting for you, that the register is accessed by the [ARM stub](https://github.com/raspberrypi/tools/blob/master/armstubs/armstub7.S#L69), which is executed before startup.S.

koparasy, 2017-06-16:
Basically in this piece of code he is writing to [CPUECTLR](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0500g/BABGHHEJ.html) not CPUACTLR. Anyway, thank you for your time.

closed 2017-06-18 by rsta2