rsta2/circle #46: Buffer Size being ignored in CSocket::Receive?

MaPePeR, 2017-09-02:
To me it looks like the buffer size is pretty much ignored when calling [`CSocket::Receive`](https://github.com/rsta2/circle/blob/master/lib/net/socket.cpp#L204). The only thing it does is overwrite the number of received bytes in case more than `nLength` bytes were received?

So i cannot really trust the number of received bytes and also cannot trust receive to not write beyond the provided `pBuffer`-Pointer?

Is this intentional? Am I missing something here?

rsta2, 2017-09-03:
You should set *nLength* to *FRAME_BUFFER_SIZE* and provide a buffer of that size as it is noted [here](https://github.com/rsta2/circle/blob/master/include/circle/net/socket.h#L73). Otherwise data, which has been received by the lower layers, may get lost in *CSocket:Receive()*. Circle will not write beyond the provided buffer, because the *memcpy()* operation uses the corrected resulting length [here](https://github.com/rsta2/circle/blob/master/lib/net/socket.cpp#L230).

MaPePeR, 2017-09-03:
Ah. I didn't notice `pBuffer` is not `m_pBuffer`. Thanks for the explanation!
Sorry for being wrong and asking dumb questions.

Awesome project!

closed 2017-09-03 by MaPePeR

rsta2, 2017-09-03:
No problem! Thanks!