rsta2/circle #47: Compiler IRQ/FIQ-Optimization

MaPePeR, 2017-09-10:
When doing my research on FIQ interrupt handlers i found [some information](https://gcc.gnu.org/onlinedocs/gcc/ARM-Function-Attributes.html), that one can use an ` __attribute__ ((interrupt ("FIQ")))` or `__attribute__ ((interrupt ("IRQ")))` to tell the compiler that a function should be used as an interrupt handler. 
I don't see any information or usage of this in circle. Is there a special reason for this?

vanvught, 2017-09-10:
The __attribute__ ((interrupt ("FIQ"))) or __attribute__ ((interrupt ("IRQ"))) is used when your code is directly called from the CPU interrupt vector. The compiler generates the register saves/restores based on the interrupt type The same code as generated in https://github.com/rsta2/circle/blob/master/lib/exceptionstub.S with lines; 61,62 and 64 for IRQ. In line 63 there is the call to the C code. Hence the C code is omitting this compiler extension. 
Moreover, using these kind of compiler extensions makes your code less portable. Hence your code might be restricted to use with specific compilers only.

rsta2, 2017-09-10:
There are often different ways to implement such things. I prefer to use assembler code for hardware stubs like the [IRQ](https://github.com/rsta2/circle/blob/master/lib/exceptionstub.S#L60) and [FIQ](https://github.com/rsta2/circle/blob/master/lib/exceptionstub.S#L70) entry and exit code, because it's more flexible.

MaPePeR, 2017-09-10:
Thanks for the response!

closed 2017-09-10 by MaPePeR