rsta2/circle #48: Why do we need a Barrier on GPIO-Output?

MaPePeR, 2017-09-10:
In [`CGPIOPin::Write`](https://github.com/rsta2/circle/blob/master/lib/gpiopin.cpp#L125-L139) you use `PeripheralEntry` and `PeripheralExit` to protect the memory-write that toggles the GPIO-Pin. 
Is that really necessary? Because on "Real-Linux" one can change the GPIO state with a simple memory write without needing any special barriers? Or are the barriers provided by the Linux kernel in that case and hidden away from the C code running on Linux? (I would be quite surprised about that, though.)

vanvught, 2017-09-10:
The answer for why we have to use barriers can be found in here -> [https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf)

I am not sure what your context is for on "Real-Linux". As we are using baremetal programming here, then any comparison with Linux is not relevant. 

When addressing the GPIO addresses from Linux userland then it is still advised to use barriers.  You can use a compiler extension for example '__sync_synchronize()' for implementing the barrier.

MaPePeR, 2017-09-10:
Thanks for your response.
I've read that document and was a little bit surprised that a lot of Linux-C-Examples don't bother to do the synchronization, but in baremetal programming its being done. 

Reading that segment on barriers again (I assume you mean "1.3 Peripheral access precautions for correct memory ordering" on page 7):

>It is theoretical possible for writes to go ‘wrong’ but that is far more difficult to achieve. The AXI system
makes sure the data always arrives in-order at its intended destination. So:
> ```
> *pointer_to_peripheral_a = value_a;
> *pointer_to_peripheral_b = value_b;
> ```
> will always give the expected result. The only time write data can arrive out-of-order is if two different
peripherals are connected to the same external equipment

So if 

```
*pointer_to_peripheral_a = value_a;
*pointer_to_peripheral_b = value_b;
```
ALWAYS give the expected result we don't need the barriers for writing to GPIO at all?

vanvught, 2017-09-10:
This is important
> It is only when switching from one peripheral to another that data can arrive out-of-order. The simplest way to make sure that data is processed in-order is to place a memory barrier instruction at critical
> positions in the code.

and

> As interrupts can appear anywhere in the code so you should safeguard those. If an interrupt
> routine reads from a peripheral the routine should start with a memory read barrier. If an
> interrupt routine writes to a peripheral the routine should end with a memory write barrier.

With 
> Linux-C-Examples

I assume that you are referring to Linux userland examples. Again, it is best practice to use barriers at any time. For a good userland C example, please let me refer to -> http://www.airspayce.com/mikem/bcm2835/
When you look at Linux drivers (which is basically baremetal programming) there are the barriers implemented.

rsta2, 2017-09-10:
The barriers are required in the document mentioned by @vanvught. for the RPi 1. I had some strange behaviour when omitting the barriers in some code. Because I cannot know the context from which *CGPIOPin::Write()* is called in user code I had to use these barriers here, to be sure.

I suppose the RPi 2 and 3 do not need these barriers. That's why *PeripheralEntry* and *PeripheralExit* do nothing there.

I think Linux is not using these barriers everywhere, but Linux has a huge testing base to sort out problems caused because of this. Circle does not have this, so I have to be more conservative while coding.

MaPePeR, 2017-09-10:
Didn't spot that these functions are NOPs for RPi 2 and 3. Thanks for answering my questions!

closed 2017-09-10 by MaPePeR