rsta2/circle #49: how to use CSPIMaster together with CSPIMasterDMA

etet100, 2017-09-12:
Hi. Sorry for my bad english.

I would like to use CSPIMaster together with CSPIMasterDMA. I want to send few bytes synchronously (CSPIMaster) than much more bytes asynchronously (CSPIMasterDMA). I manager to do this by changing

```
	write32 (ARM_SPI0_CS,   (read32 (ARM_SPI0_CS) & ~CS_CS)
			      | (nChipSelect << CS_CS__SHIFT)
			      | CS_CLEAR_RX | CS_CLEAR_TX
			      | CS_TA);

```
to

```
	write32 (ARM_SPI0_CS,   (read32 (ARM_SPI0_CS) & ~CS_CS **& ~CS_DMAEN & ~CS_ADCS**)
			      | (nChipSelect << CS_CS__SHIFT)
			      | CS_CLEAR_RX | CS_CLEAR_TX
			      | CS_TA);
```

Do you think it's correct way to do this? Using DMA and callbacks for sending 1 or 2 bytes is very inconvienent. Should i expect any issues caused by switching between DMA and synchronous mode ?

Thank you.

rsta2, 2017-09-12:
I understand the problem. Unfortunately the classes *CSPIMaster* and *CSPIMasterDMA* cannot be used together. I would suggest to implement your own SPI master class based on *CSPIMasterDMA* and to add the *WriteRead()* method from *CSPIMaster* to that class. Code using *m_SpinLock* and *m_nCSHoldTime* can be left out probably. Synchronous transfers can be done using this new *WriteRead()* method then.

etet100, 2017-09-12:
Thanks. I will do it this way.

closed 2017-09-12 by etet100

rsta2, 2017-10-12:
*CSPIMasterDMA::WriteReadSync()* has been added to the *develop* branch and will be on the *master* branch, when Circle 32 is ready. This method can be used now for the purpose you have described. Unfortunately there was one more patch required, compared to the things I wrote above, to get it running.