rsta2/circle #50: How can I wait for vertical blank?

MagnusRunesson, 2017-09-14:
Hi,

I hope this is the correct way to ask a question like this.

I am making games using this library and I would like to be able to have the game wait for the vertical sync, to avoid tearing. I've read up a bit on the subject and found two potential ways to do this.

1) Use the Set virtual offset tag, as discussed here. < https://www.raspberrypi.org/forums/viewtopic.php?t=19073 > So I've tried calling CBcmFrameBuffer::SetVirtualOffset and it returns with no error (TRUE) but the game is still not waiting for VSync.

So then I found this.

2) Enable fake_vsync_irq in config.txt and listen to IRQ 48, as discussed here < https://github.com/raspberrypi/firmware/issues/67 > So I tried connecting an IRQ handler like so:
m_Interrupt.ConnectIRQ( ARM_IRQ_SMI, IRQParty, (void*)0 );
But for some reason that function never returns. If I add logging before and after that function call I see that the log message before the call is printed but not the message after.

My IRQ handler looks like this:

volatile uint32 testValue;

void IRQParty( void* _pData )
{
	testValue++;
	write32( 0x20600000, 0 );
}

Soooooo I guess my question is, how can I wait for vsync using the Circle library?

Thank you very much for an awesome library. Super easy to use and very lightweight! My game is running within a second after the power switch is flipped, rather than first having to wait for ages for Raspbian to boot up. That is fantastic!

rsta2, 2017-09-14:
The second method works well. I have tested it with the attached test program on a RPi A+, 2, 3 and Zero.

I guess you are using a RPi 2 or 3? The address for resetting the SMI is 0x3F600000 there, not 0x20600000 like on the RPi 1. That's why your SMI IRQ handler is called all the time and the second log message cannot be displayed.

Thank you for your words about Circle!

[vsynctest.zip](https://github.com/rsta2/circle/files/1303350/vsynctest.zip)

MagnusRunesson, 2017-09-15:
Wow! Ueah, im on a rpi 3 and your works beautifully! Thank you very much for the help!

closed 2017-09-15 by MagnusRunesson