rsta2/circle #52: PWM audio on the RPI Zero

EvilDraggie, 2017-10-20:
I would like to use the PWMSounddevice with the RPI Zero but this requires the standard pins (40,etc) to be changed because the Zero doesn't have those audio pins available. I did see you had a quick solution to this in the Oct 1, 2017 commit, but this does not seem to work (anymore).

Any pointers on how this can be done currently?

MaPePeR, 2017-10-20:
I think the solution does not work out of the box. You need additional hardware to turn these pwm signals into something worth listening to, i think.

rsta2, 2017-10-20:
I have tested PWM audio on the RPi Zero again with the sources from *master* branch with *sample/12-pwmsound*. It is working well.

As @MaPePeR wrote, you need some external circuit to use PWM support on the RPi Zero. [This one](https://learn.adafruit.com/adding-basic-audio-ouput-to-raspberry-pi-zero/overview) should work. Please note that Adafuit is using different pins for the output, Circle is using GPIO12 and GPIO13 (SoC numbering).

The PWM support for RPi Zero has to be enabled in *include/circle/sysconfig.h* before building. Remove the comment from the *USE_PWM_AUDIO_ON_ZERO* option in that file!

EvilDraggie, 2017-10-20:
I am aware of the additional circuitry (been building synthesizers with Arduino's so far, also utilizing PWM) and that's all good. I tested the output with a modified version of the PWMoutput sample on GPIO18.

I shall try again with GPIO pins 12 & 13, I don't think I've tried those in combination with the PWMsound sample.

However there's no USE_PWM_AUDIO_ON_ZERO line in the sysconfig.h file, nor that extra bit of code in machineinfo.cpp (I'm using the most recent branch too)?

EvilDraggie, 2017-10-20:
Okay, checked the master branch, I think I somehow got the wrong branch. Really not sure how haha

EvilDraggie, 2017-10-20:
Okay, after changing the mentioned things and rebuilding everything it is now working. Nice tune :D

Thanks for the quick replies and props on the work you have done so far!

rsta2, 2017-10-21:
OK and thanks!

closed 2017-10-21 by rsta2