rsta2/circle #53: POSIX socket compatibility possible?

MaPePeR, 2017-10-27:
I'm just curious: Would it be possible to make the circle socket api look like normal sockets, so one can use a default C library that uses sockets and it would work out of the box?

Probably a lot of work, though.

rsta2, 2017-10-28:
Maybe not that much work for a basic subset of the POSIX socket features, which can be realized using Circle. Circle does support the AF_INET address family only and only a subset of functions and only very few socket options. But this could be extended in some way, if there is some need for it.

Providing POSIX sockets makes probably only sense with a C standard library implementation available. @smuehlst is currently working on porting newlib onto Circle. With the right toolchain his very interesting [project](https://github.com/smuehlst/circle-stdlib) will allow to use the features of the C++ standard library too. This will make porting applications onto Circle much easier.

I thought about, if this POSIX socket support could be implemented as a separate library project to be included as a sub-module into the project of @smuehlst. But because the file functions in newlib and the socket functions use the same file descriptor pool and because there are functions which are used in both domains (e.g. close()) this might be not doable.

I have seen that there is some socket support in newlib (e.g. for Linux targets). I don't know how difficult it is, to port this to Circle. Perhaps @smuehlst is reading this and can answer this question.

MaPePeR, 2017-10-28:
For a c standard library in circle this comes to my mind: https://www.fefe.de/dietlibc/
I don't really know anything about that, but from my basic understanding it might be a very good choice for circle?

vanvught, 2017-10-28:
Using newlib is a very good choice. It is well documented and it is available with most Linux distributions. For the latter, there is no need to build your own toolchain with newlib support. Of course when you are eager to build newlib with the special flags for the Raspberry Pi ARM version, then you can build your own environment -> http://www.raspberrypi-dmx.org/raspberry-pi-dmx512-rdm/development-environment

rsta2, 2017-10-28:
newlib has system call stubs too, which have to be implemented to use the full features. @smuehlst has already done this to some degree. Many functions are already working, including features of the C++ standard library. I'm sure he will get this running and there is no need for a different code base.

smuehlst, 2017-10-28:
newlib indeed has support for socket() for some platforms, so I agree with @rsta2 that it could be feasible to expose Circle's networking features through the POSIX networking interface. I will need to do more research on how that could be implemented.

MaPePeR, 2017-10-28:
Ah. i didn't really know about newlib and misunderstood it. (I thought this was a working title for something you were working on 😅) Sounds like its probably a much better choice than dietlibc - especially if you already started on making it work.

Sticking to existing/ancient interfaces could make it a lot easier to use code with circle - but i don't know if that is a real requirement at all, so POSIX-sockets was just a random thought i had.

rsta2, 2017-10-29:
Using the C and/or C++ standard libraries can make bare metal programming and especially porting existing applications much easier. That's why I'm supporting this project with Circle. Adding POSIX sockets to this is a good idea. So thanks for that!

rsta2, 2024-02-06:
I think, this is discussed in smuehlst/circle-stdlib#31 now. So this can be closed.

closed 2024-02-06 by rsta2