rsta2/circle #54: CDeviceNameService - list devices

etet100, 2017-11-13:
Hi. I'm trying to run file access example. It keeps saying that "partition not found". I wonder why there is no ListDevices method in CDeviceNameService. It could help me to find correct device name.

Andrzej

rsta2, 2017-11-13:
Hi Andrzej, I have added the suggested method. It can be called using

``m_DeviceNameService.ListDevices (&m_Screen);``

from *CKernel::Run()*. Block devices are preceded with a 'b' character. Thanks for this suggestion.

rsta2, 2017-11-13:
I forgot to mention: it's on the *develop* branch.

etet100, 2017-11-14:
Thanks. Unfortunately i can't link circle develop (and circle-newlib) since you broke backward compatibility. I have to mamy libraries compiled with float-abi=hard.

rsta2, 2017-11-14:
Can you please try to change the ARCH options in *Rules.mk* for your RASPPI model back to -mfloat-abi=hard ? Does it work then? This is still under development and would be valuable information. Perhaps another way would be to recompile all the libraries, if this is possible.

The problem is, that for the Linaro GCC 7.1.1 toolchain there isn't support for -mfloat-abi=hard for the RPi 1 any more. That's why I changed this setting, but perhaps I have to revert it or make it configurable.

rsta2, 2017-11-14:
I have added a build option *FLOAT_ABI* to make it configurable. It defaults to the previous setting *hard*, but may be set to *softfp* if required. I think, this is the best solution.

etet100, 2017-11-14:
Hi. Thanks. I'm sure this setting is fine.

But i still can't link it with my libraries. Plenty of stupid errrors like 

..\circle-stdlib\install\arm-none-circle\lib\libm.a(lib_a-w_acos.o)||In function `acos':|
w_acos.c:(.text+0xb4)||undefined reference to `__errno'|
..\circle-stdlib\install\arm-none-circle\lib\libm.a(lib_a-w_sqrt.o)||In function `sqrt':|
w_sqrt.c:(.text+0xc8)||undefined reference to `__errno'|

Damn, i'm hopeless.

rsta2, 2017-11-15:
I have to apologise, because I invited you to do tests of versions from our development branches. Previously this should not be a problem, because versions, checked in into the Circle *develop* branch, have passed a module test and should work, normally. But we have a new situation now, that we are working together on different projects and sub-projects, using this branch for versions, which are not tested very well, to be able to communicate. This was not clear to me.

Because we had several commits yesterday, I do not know, which commits of the different projects you have used for your test and cannot reproduce your problem. I would say, it's better to sit back now and give us some days to sort this out and to come to a state, which makes sense for further testing. I will inform you here, when this has been reached.

smuehlst, 2017-11-16:
@etet100: Apparently you are building Circle in the context of the [circle-stdlib](https://github.com/smuehlst/circle-stdlib) project. There was an error in the Makefiles of the samples that the C library libc.a was put before the math library libm.a on the linker command line. This order was wrong because libm.a references symbols from libc.a, namely __errno.

This is fixed now, the error should no longer appear starting with commit smuehlst/circle-stdlib@a1ed3f79fa3dfff001638424c918da610aed50ec.

etet100, 2017-11-18:
Thank you very much! A found some more bugs in my code and now it's working fine.

rsta2, 2017-11-19:
Great! Good to know, that it is already working fine. So I think, this can be closed.

closed 2017-11-19 by rsta2