rsta2/circle #55: Both Circle and Circle64 are missing CursorLeft support

am-roberts, 2017-11-22:
Both Circle and Circle64 have the code for CursorLeft missing in lib/screen.cpp

lib/screen.cpp
```
void CScreenDevice::Write (char chChar)
{
...
        case ScreenStateBracket:
...
                case 'C':
                        CursorRight ();
                        m_nState = ScreenStateStart;
                        break;

                case 'D': // Added by AMR, missing in original
                        CursorLeft ();
                        m_nState = ScreenStateStart;
                        break;

                case 'H':
                        CursorHome ();
                        m_nState = ScreenStateStart;
                        break;
...
}
```

rsta2, 2017-11-23:
This "\x1b[D" sequence for cursor left is not supported, that's right. But "\b" (^H) does the same in Circle. Historically the Circle screen driver sequences were taken from the VT100 terminal, which uses ^H for cursor left on output. I think "\x1b[D" comes from ANSI and can be supported, if needed. I will add this to the next Circle release.

rsta2, 2017-12-02:
I have added this control sequence to Circle and Circle64. It's on the *develop* branch of Circle and will be on *master* when Circle 34 is ready.

closed 2018-01-16 by rsta2