rsta2/circle #56: Raspberry zero - speed settings

etet100, 2017-11-26:
Hi. As you probably know the default speed settings for raspberry zero are slightly higher than settings for raspberry 1. Since you use fixed values for both raspberry 1 and zero i had to make some modifications in circle core to change speed settings:

1) make CCPUThrottle::SetClockRate public accessible

rsta2, 2017-11-26:
These modifications are not really needed to run the RPi Zero at 1 GHz. It's not true, that I use fixed values for the allowed range of the ARM clock rate. *m_nMaxClockRate* is first set to 600 in the constructor of class *CCPUThrottle* but then [overwritten with the actual value requested from the firmware](https://github.com/rsta2/circle/blob/19135be469e2c9a249e73adfab738d36ca6d4b51/lib/cputhrottle.cpp#L61). That is 1000 for the RPi Zero.

You only need to include the *fast=true* option in *cmdline.txt* or include the member initialiser `m_CPUThrottle (CPUSpeedMaximum)` in *CKernel*. Your RPi Zero will run at 1 GHz then. I have tested this again with *sample/26-cpustress*.

I think, the firmware knows best on their own, how to handle turbo mode and voltage settings for different clock rates. I remember, that implementing this CPU clock support was very tricky, so please be careful!

etet100, 2017-11-26:
Thank you for explanation. I forgot about cmdline setting. 

However, i should mention what i'm trying to achieve. I need to speed up SPI a little bit. Current speed is to low to send 60 frames per second, but next step means twice higher speed, which is way to fast for my LCD. The only solution i found is to speed up core clock or speed up SPI and slow down core clock. This is why i can't use any predefined speed settings.

rsta2, 2017-11-26:
I have no experiences with overclocking by myself. If you really need this for your application, I would suggest doing it using the settings in [config.txt](https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md).

rsta2, 2017-11-29:
After thinking about this issue again, I decided, that I will not support these overclocking (or underclocking) settings in Circle directly. There are several reasons for this:

* Changing the clock settings in an arbitrary way may be risky.
* Clock settings can be manipulated more safely in config.txt, because it is checked by the firmware. This is the recommended way to do it.
* Your requirements are very special. I don't think, that many people would use such support.
* If you really need these functions, you can implement this by yourself (on your own risk), as you have started it already.

This issue can be closed soon.

etet100, 2017-11-30:
Hi. You're right. Thanks.

closed 2017-11-30 by etet100