rsta2/circle #57: Raspberry pi 3 hanging on serial print

Matheus-Garbelini, 2017-12-17:
Hello, I'm having a problem while testing with raspberry pi 3.
The blink example works and i can see the led blinking, but as soon i insert a serial code, the rasp hangs, also if i try to run other examples i receive nothing at the serial. Could it be a different bootcode problem? If so, could you post the boot binaries tou used to perform test with raspberry pi 3? Thanks

```C++
//
// kernel.cpp
//
// Circle - A C++ bare metal environment for Raspberry Pi
// Copyright (C) 2014-2016  R. Stange <rsta2@o2online.de>
// 
// This program is free software: you can redistribute it and/or modify
// it under the terms of the GNU General Public License as published by
// the Free Software Foundation, either version 3 of the License, or
// (at your option) any later version.
//
// This program is distributed in the hope that it will be useful,
// but WITHOUT ANY WARRANTY; without even the implied warranty of
// MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
// GNU General Public License for more details.
//
// You should have received a copy of the GNU General Public License
// along with this program.  If not, see <http://www.gnu.org/licenses/>.
//
#include "kernel.h"
#include <circle/gpiopin.h>
#include <circle/timer.h>

CKernel::CKernel (void)
{
}

CKernel::~CKernel (void)
{
}

boolean CKernel::Initialize (void)
{
	return TRUE;
}

TShutdownMode CKernel::Run (void)
{
	CGPIOPin AudioLeft (41, GPIOModeOutput);
	CGPIOPin AudioRight (40, GPIOModeOutput);

	// flash the Act LED 10 times and click on audio (3.5mm headphone jack)
	for (unsigned i = 1; i <= 10; i++)
	{
		m_ActLED.On ();
		AudioLeft.Invert ();
		AudioRight.Invert ();
		CTimer::SimpleMsDelay (200);

		m_ActLED.Off ();
		CTimer::SimpleMsDelay (500);
	}

	return ShutdownReboot;
}

```

rsta2, 2017-12-17:
You only need to go to the Circle boot/ directory and do `make` there to get the most recent firmware from [here](https://github.com/raspberrypi/firmware/tree/master/boot). Please copy the files *bootcode.bin*, *start.elf* and *fixup.dat* to a uSD card along with the kernel image file.

If you want to get output to the serial interface on GPIO 14/15, you need to create a file *cmdline.txt* on the uSD card, which has this contents:

``
logdev=ttyS1
``

I append a kernel image of *sample/04-timer*, which has been successfully tested with the firmware of today on a RPi 3 v1.2.

[kernel8-32.img.zip](https://github.com/rsta2/circle/files/1565460/kernel8-32.img.zip)

Matheus-Garbelini, 2017-12-17:
Thank you a lot. It was the missing cmdline.txt. it worked like a charm.
Also, i tested the fractal, mandrelset sample, and it worked very well. Altough, when reading the code from kernel.cpp

it initializes mandrelset class like this:

```C++
TShutdownMode CKernel::Run (void)
{
	m_Logger.Write (FromKernel, LogNotice, "Compile time: " __DATE__ " " __TIME__);

	m_Mandelbrot.Run (0);

	return ShutdownHalt;
}
```

But, if mandrelset receives an first argument that corresponds to each core number, how this constant "zero" be different for each core instance?

rsta2, 2017-12-17:
There is a different initialization for core 0 and core 1-3. Core 0 is started at 0x8000 by the firmware and is directed to CKernel::Initialize() and then to CKernel::Run() by the Circle sysinit code. The secondary cores 1-3 are parked in a loop after boot by the firmware and will be released from that loop in CMultiCoreSupport::Initialize() which is called in CKernel::Initialize(), because the class CMandelbrotCalculator is derived from CMultiCoreSupport.

The secondary cores then jump to [CMultiCoreSupport::EntrySecondary](https://github.com/rsta2/circle/blob/master/lib/multicore.cpp#L170) and in line 185 to this Run() routine for each secondary core. Because core 0 has its own initialization path, it explicitly calls CMandelbrotCalculator::Run() with its core number 0. So the execution of all cores reaches CMandelbrotCalculator::Run() in the end.

Matheus-Garbelini, 2017-12-18:
Thank you so much, i'm working with circle to provide a wrap around arduino. I've already manage to modify a bootloader from dwelch67 and included a python script to automate flashing (Takes me around just 1 sec for flashing the 01-blink example with a baud rate of 1M). I'm looking forward to contribute with this project. As a windows user, I wouldn't be more happier to see this project running out of the box with arduino IDE.

closed 2017-12-18 by Matheus-Garbelini