rsta2/circle #59: Timer interrupt in FIQ

Matheus-Garbelini, 2017-12-19:
Can we use timer interrupt in FIQ instead of IRQ?
I'm tring to make an application that generates some sort of wave in pwmaudio output.

Looking through the code i can see that tiner initialization connects only to irq and uses arm_timer3 (there are more than 2 timers on rasp?). Also, if i'm using multicore is it possible to generate a timer interrupt in each one? Thanks.

rsta2, 2017-12-19:
Yes, Circle uses the ARM timer 3 only. Timer 0 and 2 are used by the VPU, but timer 1 is free. I know of applications which are using it for their own purposes. That's why Circle will not use it and you could use it. You can also connect it to the FIQ, if you want. Perhaps you can get the information to use the timer 1 from *lib/timer.cpp*. Otherwise I can give you further advise.

The Circle timer handler uses the IRQ, because the FIQ should remain free for user purposes. All IRQs and FIQ are handled by core 0.

Matheus-Garbelini, 2017-12-19:
@rsta2. Thank you, its working very well  up to 2us @1.3Ghz. My god, so much faster than arduino,esp32 or cortex m7. And we have 4 main loops to play xD

closed 2017-12-19 by Matheus-Garbelini