rsta2/circle #60: Config.txt over_voltage option

koparasy, 2017-12-20:
This is more like a question and not an actual issue. I am trying to reduce the power consumption of a raspberry pi 3b and i am using Circle. To reduce the power consumption i am also trying to use the over_voltage option in the config.txt file. As explained in ([here](https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md)) if you set negative values to the option the raspberry pi uses less supply voltage for the CPU.  So I am booting the PI with this option set to -8. I am not seeing any differences in the power consumption though. Should I somehow set this value during boot time?

Matheus-Garbelini, 2017-12-20:
@koparasy I not chanced the values of over voltagem yet, but when i used arm_freq_min to 100Mhz using pi3, the consumptuon dropped a lot.

rsta2, 2017-12-21:
I didn't experiment with voltage settings, so I cannot give advise here. But I guess, reducing the clock(s) will help more to reduce power consumption, as @Matheus-Garbelini did confirm.

closed 2018-01-01 by rsta2