rsta2/circle #61: Support for Screen DMA

Matheus-Garbelini, 2017-12-21:
Hello, i'm porting the littlevGL GUI library to circle and is working well, but i think it can be speed optimized if using some sort of write to screen trough DMA. The library is a huge improvement over ugui, more easier and powerful (Even has pc simulator). 
Take a look: http://www.gl.littlev.hu

Do you know if its possible to support this in circle without breaking the screen class? Thanks, i'm looking forward to it.

rsta2, 2017-12-22:
Generally it should be possible to transfer image data to the screen using DMA. But the CDMAChannel class currently does not support the 2D Mode, which is probably needed. I can add this as a feature request to my TODO list, but I am working on other things currently.

There can be multiple writers to the frame buffer memory, so there shouldn't be a problem with the CScreenDevice class, but of course if there is some output from the GUI library and from CScreenDevice too, the output will be mixed. CScreenDevice::Write() will not write into a window or something similar.

Matheus-Garbelini, 2017-12-22:
Thank you a lot, i will try to have a look into this also. Here's the new gui that i will be merging in my branch soon:

[image]

rsta2, 2018-01-01:
This GUI looks good. But I'm not sure, if Circle needs another GUI library. Have you thought about providing this in your own fork of the Circle repo?

Matheus-Garbelini, 2018-01-01:
@rsta2 offcourse. I said my branch but i meant my fork.
It's a wonderful gui but it would be redundant in oficial circle. It would only make sence to be merged if this library would replace the current ugui library.
PS: Happy new year.

rsta2, 2018-01-02:
Replacing the uGUI library would be difficult, because there are existing applications, which are using it. Happy new year!

rsta2, 2018-01-26:
I have updated the class *CDMAChannel* so that it supports the 2D mode now for memory copy. It should be ready now to be used for flushing pixel colour information from a virtual screen buffer to the screen. The appended sample program shows how to use *CDMAChannel* for that purpose.

[screendmatest.zip](https://github.com/rsta2/circle/files/1667618/screendmatest.zip)

Matheus-Garbelini, 2018-01-26:
@rsta2 this is awesome. I will with the GUI library and give you a feedback. Thanks 👍

rsta2, 2018-01-26:
Thanks. According to the [µGUI Reference Guide](http://embeddedlightning.com/download/reference-guide/) hardware acceleration supports "Draw Line" and "Fill Frame" functions only. There is no virtual display buffering. Unfortunately that's why this should not help here.

Matheus-Garbelini, 2018-01-29:
@rsta2 the dma channel worked well, it's really fast now. Event in 100Mhz, there's no slow down in frames.
Unfortunatelly, It seems I can't use SetCompletionRoutine for interrupt. It seems the interrupt function is never called.

rsta2, 2018-01-29:
There is a modified sample program attached to this comment, which works with interrupt. Be sure to give a pointer to *m_Interrupt* as second parameter to the constructor of the DMA channel object.

[screendmatest.zip](https://github.com/rsta2/circle/files/1673304/screendmatest.zip)

rsta2, 2018-03-21:
@Matheus-Garbelini Are there any news on this? I'm planning a new release in April and do collecting the things, which have to be done until then.

rsta2, 2018-03-28:
The 2D mode support in the class *CDMAChannel* is on the *master* branch now. I suppose it is working well, even with interrupt.

closed 2018-03-28 by rsta2

Matheus-Garbelini, 2018-03-28:
@rsta2 sorry for the delay. So, I've not worked something new yet, but the last time i tested cdmachannel it was ok.

rsta2, 2018-03-29:
@Matheus-Garbelini Good to know, that it is working for you. Thanks for testing!