rsta2/circle #62: Assert on synchronizationevent.cpp

jsanchezv, 2018-01-02:
Hi,

The last days I'm fighting with a random problem in synchronizationevent.cpp. Sometimes, the program aborts in the line:

assert (m_bState)

(line 65, Wait method).

After many tests, I suspect that sometimes different CPUs are watching different values for m_bState. My program calls to Set method from CPU-1 and Clear/Wait from CPU-0. Yesterday, I've put a call to DataSyncBarrier() after setting m_bState in Set & Clear methods. I have the program running the last 24h without any problem.

The bug is hard to reproduce. Sometimes lasts 10 mins, sometimes 3h. The variable is declared as volatile as expected, so I only can think in a difference between CPU caches or so.

My program can't run on ARMv6 models, so I'm not tested these boards. Probably, don't need the call to DataSyncBarrier().

Best regards
José Luis

rsta2, 2018-01-02:
This problem arises with the scheduler opened for multi-core usage. I think, your suggested fix is the right one. The DataSynchBarrier() ensures that the update to m_bState is visible to all cores. I have added the barriers as suggested to the new branch *scheduler-revised*. This branch contains more updates to allow tasks to run on secondary cores (1, 2 or 3) too. Thanks for reporting this issue, suggesting a fix and testing it!

rsta2, 2018-01-12:
The new multi-core aware scheduler has been withdrawn for the next Circle release because of problems with it (e.g. increased latency). There is currently also no known application, which would use it. In this situation it does not make sense to change the Circle scheduler model, which allows tasks to run on a single core only (normally core 0), which is valid for a longer time and tested well.

Without the new scheduler and with all tasks running on a single core the assert failure which was the reason for this issue should not arise any more and does not need to be fixed.

jsanchezv, 2018-01-12:
The assert problem arises with old scheduler too because Set method is called from CPU-1 at any case and the change needs to be seen by Wait in CPU-0.

rsta2, 2018-01-12:
I thought this scenario was in a version of ZXBaremulator, which you do not want to release. Normally it is not allowed to use *Set* and *Wait* from different cores with the old scheduler and my problem is, to give this free for an application which is not really available. For a framework like Circle I have to keep an eye on clear models for the different components.

jsanchezv, 2018-01-12:
Yes, I've found the problem coding ZXBaremulator that I can't release now
you knows why. The fact is that calling Set from a CPU different from 0
works, and this allows synchronization between CPUs, impossible at any
other form.

But include the patch is entirely your decision and I can do nothing about
this.

jsanchezv, 2018-01-12:
Yes, I've found the problem coding ZXBaremulator, that I can't release now you knows why.

The fact is that calling Set from a CPU different from 0 works, and this allows synchronization between CPUs, impossible at any other form.

But include the patch is entirely your decision and I can do nothing about this.

rsta2, 2018-01-14:
I have added this patch to the *develop* branch, so it will be part of the next release. The new beta version of ZXBaremulator is working well.

jsanchezv, 2018-01-14:
Thanks Rene.

closed 2018-01-16 by rsta2