rsta2/circle #64: Support for SPI1

Matheus-Garbelini, 2018-01-19:
Hello, I've seen that spi0 is supported, can spi1 be supported in Circle? It's just a matter of changing some register to make it work without DMA or raspberry is using spi1 internally for other things? Thanks

rsta2, 2018-01-19:
I don't think SPI1 is used internally. @vanvught has pointed me on this some time ago and he has implemented a [working driver](https://github.com/vanvught/rpidmx512/blob/master/lib-bcm2835/src/bcm2835_aux_spi.c). I will try to port it to Circle.

rsta2, 2018-01-24:
The SPI1 driver is ready. It is on the *develop* branch at the moment and is implemented in the new class *CSPIMasterAUX* in *include/circle/spimasteraux.h* and *lib/spimasteraux.cpp*. I have updated the *sample/23-spisimple* so that it can be used with this driver. Please have a look at the *README* file in this directory for details!

Matheus-Garbelini, 2018-01-24:
@rsta2 thanks. I will test this feature and let you know 👍

closed 2018-01-24 by Matheus-Garbelini