rsta2/circle #66: Reading files from the SD card that boots the RPi

tohyf, 2018-02-04:
I saw in the README that "EMMC SD card interface driver" is supported, but I can't find any sample code that demonstrates how to read files from the SD card. So my question is, can I read files from the FAT partition from the SD card, like how the files are read from USB device in the "15-files" sample code?

MagnusRunesson, 2018-02-04:
There are more samples in the addon folder. Reading from the SD card is done here. 

https://github.com/rsta2/circle/tree/master/addon/SDCard/sample

rsta2, 2018-02-04:
The SD card driver and sample is in the *addon/* subdirectory as @MagnusRunesson writes, because the SD card driver has been ported from [here](https://github.com/jncronin/rpi-boot/blob/master/emmc.c). Most code ported from external projects is placed in this subdirectory.

closed 2018-02-09 by rsta2