rsta2/circle #67: libgcc.a linking issue

tohyf, 2018-02-04:
The code compiles fine when using ./makeall --nosample. However, when it tries to compile any sample, I would get undefined reference to __aeabi_uidiv and similar functions, and the output error looks like this:

arm-none-eabi-ld -o kernel.elf -Map kernel.map -T ../../circle.ld ../../lib/startup.o main.o kernel.o oscillator.o  ../../lib/sched/libsched.a ../../lib/libcircle.a 
oscillator.o: In function 'COscillator::NextSample()':
..../circle/sample/34-sounddevices/oscillator.cpp:129: undefined reference to '__aeabi_uidiv'
../../lib/libcircle.a(timer.o): In function 'CTimer::TuneMsDelay()':
....../circle/lib/timer.cpp:446: undefined reference to '__aeabi_uidiv'
../../lib/libcircle.a(screen.o): In function 'CScreenDevice::GetColumns() const':
......./circle/lib/screen.cpp:139: undefined reference to '__aeabi_uidiv'
< and a lot more >

My guess is that it can't find the libgcc.a properly, so as a workaround I added the path of libgcc.a to the LD command in the $(TARGET).img section (I used /opt/gcc-arm-none-eabi-7-2017-q4-major/lib/gcc/arm-none-eabi/7.2.1/libgcc.a), and it seems to compile fine.

As there are quite a few libgcc.a files in the toolchain folder, which one is the correct libgcc.a to use? Did I installed my toolchain incorrectly? I am using Ubuntu 14.04 and I simply added '/opt/gcc-arm-none-eabi-7-2017-q4-major/bin' to the $PATH variable (as i extracted the toolchain folder to /opt ) before I run ./makeall.

rsta2, 2018-02-04:
Normally the location of *libgcc.a* is detected automatically in *Rules.mk* and *libgcc.a* is given with its full path to the linker. I used the option `-print-file-name=libgcc.a` of the compiler to get the location [here](https://github.com/rsta2/circle/blob/master/Rules.mk#L67). This should be the following path with your toolchain for the RPi 1:

`/opt/gcc-arm-none-eabi-7-2017-q4-major/lib/gcc/arm-none-eabi/7.2.1/hard/libgcc.a`

I have downloaded this toolchain from [here](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads) and building the Circle samples works all right with it here. Normally it is enough to set the $PATH variable so that the toolchain binaries can be found.

If you have problems using *libgcc.a* you can define `STDLIB_SUPPORT = 0`in *Config.mk*. Circle samples will be built with self implemented `__aeabi_` functions then and do not need *libgcc.a* in this case.

tohyf, 2018-02-05:
Thanks for the reply. I did download and use the toolchain from that ARM website (and i got 2017-q4 version) all the while, so I am wondering if I am missing anything. I tried running that `-print-file-name` command manually and I got the same filename as you shown me, interesting.
Probably the "make" I am using is somewhat outdated ? Mine is version 3.81, comes from my distribution repo. How about yours?

tohyf, 2018-02-05:
Nevermind, I found out that the != operator is only added in Make 4.0, oops.

rsta2, 2018-02-06:
I have GNU make 4.0. I will think about a solution for this problem. Thanks for info! For the moment you should specify the *libgcc.a* path manually or use `STDLIB_SUPPORT = 0`.

rsta2, 2018-03-21:
Finally a test for the required GNU make version has been added to *Rules.mk* at the *develop* branch. It's a little cryptic, but should work. Thanks for the suggestion again.

rsta2, 2018-03-28:
The *make* version check is on the *master* branch now.

closed 2018-03-28 by rsta2