rsta2/circle #68: Serial Interrupt transmit problem

Matheus-Garbelini, 2018-04-06:
Hello, I'm trying to work with serial in asynchrounous way by using the already provided interrupt support from circle serial class.

I made an inherited class wrapping serial and I'm the problem i am having is related to not being able to send full TX messages when interrupt for serial is enabled (via circle serial class constructor support)

When i run my code without the interrupt support (the blocking way) the serial prints fine in the monitor, but as soon i add the interrupt in the constructor it prints the messages in shorts blocks only and they will not the next bytes written in tx buffer until the next serial.Write is called. For instance, if i try to print a message each second, the rasp send a part of the first message and continue to send it only after the in the next second (if a lower the delay it will print faster the serial off course).

What i tried:

- I tried doing that not using delays and the results are the same.

- Using the serial class as it's without my classes and the same appens

- Not using Logger, but if i do that, nothing appear on the serial monitor.

- For some reason, if i not call m_Logger.Initialize(pTarget); it will not initialize serial even tough it's initialized before or after, and don't let me uses the other cores if not called also.

I sent you an attachement so you can test this on a rasp 3 to verify.

The folder Arduino must be placed in the addons folder and compiled with make.
The other is the project i'm talking about, just export CIRCLE_PATH to your circle dir and it should be compile fine.

I'm trying to make a serial wrapper to be equal to arduino libraries, but without been able to get the received bytes directly from the interrupt variable that controls is i cannot do it (maybe considering to change m_nRxOutPtr to be a protected instead of private?

If you have problems to test or need some more information, let me know. Thanks.

[baremetal.zip](https://github.com/rsta2/circle/files/1884505/baremetal.zip)

rsta2, 2018-04-07:
That's really interesting, what you are doing towards an Arduino integration, as I think. Thanks for using Circle for this.

I've found two problems in *TestBoardRaspberry/main.cpp*. The *setup()* function in this file should look like this to correct the issue:

```
void setup() {

	// True setup
	m_PWMPin.SetMode(GPIOModeOutput);
	m_PWMPin.Write(1);
	delay(1000);
	m_Interrupt.Initialize();
	Serial.begin(115200);

	m_Clock0.Start(CLOCK_DIVIDER); // Negative Power supply
	//m_PWMPin.Write(0);
	m_PWMOutput.Start();
	MotorWrite(0);
	m_PWMPin.SetMode(GPIOModeAlternateFunction5);

	Serial.println("Board Initialized");

	// ----------------

	m_Logger.Initialize(&Serial);

	m_Timer.Initialize();

	//Processors.Initialize();

}
```

1. The order of calling the *Initialize()* methods is critical. I have to admit, that this can be difficult to find out. *m_Interrupt.Initialize()* must be called before *Serial.begin()*, because the serial initialization uses the *m_Interrupt* object.

2. *m_Logger.Initialize(&Serial)* initializes the logging output fixed to the Serial device. Because there is no *CScreenDevice* object in your system, this is the only option. The logger must be initialized, because *m_Timer.Initialize()* writes to it and will freeze the system if this was not done.

*m_Options.GetLogDevice()* does not need to be called in your case then, because the logging device is fixed. *m_Options.GetLogDevice()* would return the string "tty1" otherwise, if there is no file *cmdline.txt* on the SD card with the "logdev=" option in it, and *m_DeviceNameService.GetDevice()* would return a null pointer when trying to find this device, which does not exist in your case. Calling *m_Logger.Initialize()* with a null pointer will not work.

With the *setup()* function shown above everything is running well here. Even when enabling the secondary cores, I can see their serial output.

---

I can offer you to add the following protected methods to the class *CSerialDevice*:

```
	unsigned AvailableForWrite (void);
	unsigned AvailableForRead (void);
	int Peek (void);
	void Flush (void);
```

With these additional methods it should be possible to create the Arduino serial wrapper class. I would prefer this over making the member variables protected. Would this be OK for you?

Matheus-Garbelini, 2018-04-07:
@rsta 2. Thank you very much. I forgot to mention i had a cmdline.txt file poiting to ttyS1.
The solution you proposed to get the available bytes are perfect then. With this i can try to using the esp8366 at software library for arduino.

Onde more thing, i tried to use __weak declaration for the other blank loops so the user can optionally write just the main loop for core 0 even tough other processos are enabled. But it gives me errors like the toolchain don't support this.

Do you know how i can declare loop1,2,3 functions with a default implementation that could be overwritten by a user function in the main.CPP?

rsta2, 2018-04-07:
OK, I will update the *CSerialDevice* class as proposed in the next week. ESP8266 support is very interesting.

I think you have to add this to the prototypes in *Arduino.h*:

```
void setup();
int loop();
void loop2()  __attribute__ ((weak));
void loop3()  __attribute__ ((weak));
void loop4()  __attribute__ ((weak));
```

Then you can define default functions in *Arduino.cpp* like that:

```
void loop2 (void)
{
	asm volatile ("wfi");
}

void loop3 (void)
{
	asm volatile ("wfi");
}

void loop4 (void)
{
	asm volatile ("wfi");
}
```

These can be overwritten by user defined functions in *main.cpp*. I tried this with my toolchain (GCC 6.3.1) and checked the assembler code in *kernel7.lst*. I didn't test the *kernel7.img* so far.

rsta2, 2018-04-09:
I have added the proposed methods to the class *CSerialDevice* for Arduino emulation. It's currently on the *develop* branch.

Matheus-Garbelini, 2018-04-09:
@rsta2 thank you, i confirm it is working very well

Matheus-Garbelini, 2018-04-09:
Also, I'm using some extra code in serial interrupt before adding to the fifo buffer so i can reupload the code without needing to reset the board (you know the laziness right? haha)
Do you think that this idea could be optionally enabled if added a definition in Config.mk? (Maybe not the same '+' characters in sequence also).
It sort put a little more processing time in the interrupt but for development porpoises could make things more comfortable. I mean, thinking in an arduino way that the user just hits the upload button without needing to do anything else.

```C++
if (((m_nRxInPtr+1) & SERIAL_BUF_MASK) != m_nRxOutPtr)
		{
            u8 data = nDR & 0xFF;          
			m_RxBuffer[m_nRxInPtr++] = data;
			m_nRxInPtr &= SERIAL_BUF_MASK;
            if(data == '+') 
            {
                ++sequenceCount;
                if(sequenceCount==8) reboot();
            }
            else 
            {
                sequenceCount = 0;     
            }
           
		}
```

Also the bootloader will need to be passed this argument if it exists, the code initially look like that:

```python
ser = serial.Serial(
        port=flashport,
        baudrate=115200,
        parity=serial.PARITY_NONE,
        stopbits=serial.STOPBITS_ONE,
        bytesize=serial.EIGHTBITS
    )

    ser.write("++++++++")
    sys.stdout.flush()
    ser.close()
    sleep(4)
```

rsta2, 2018-04-09:
A spinlock does only affect another core, when it tries to acquire the same spinlock. So this is not a problem here, if only one core is accessing the serial interface at a time.

I think, we should keep these longer development internals out of these Issues which are more to report problems and for requesting a single new feature. I will write you an Email to the address on your GitHub page.

Matheus-Garbelini, 2018-04-09:
@rsta2 fair enough. Thanks.

closed 2018-04-09 by Matheus-Garbelini