rsta2/circle #69: Problem with last firmware & u-boot

jsanchezv, 2018-04-07:
U-Boot supports now the new PI 3B+, but I guess that by this FW issue:

[https://github.com/raspberrypi/firmware/issues/953](https://github.com/raspberrypi/firmware/issues/953)

Circle have problems on MMC initialization:

00:00:02.70 emmc: Inquiry ACMD41 failed.

U-Boot includes a patch to work out the problem (commit 8ae1f8298820 from U-Boot git repo).

rsta2, 2018-04-08:
With which firmware commit did you test this? Unfortunately I cannot reproduce it with *3aa806091dc* from Apr 6. The *addon/SDCard/sample* works well with it on both RPi 2 and 3B+. Or does this only occur when booting Circle using U-Boot?

The [U-Boot patch](https://github.com/u-boot/u-boot/commit/8ae1f8298820) uses the core clock instead of the EMMC clock for the SD card interface, but how to find out, which clock to use with the firmware which is actually installed?

The firmware issue is not closed yet, so it seems they are still working on it.

jsanchezv, 2018-04-08:
With FW from last commit in repo:

VC_BUILD_ID_TIME: 17:57:04
VC_BUILD_ID_TIME: Apr  6 2018

The problem is present when I boot using last u-boot with PI-3B+ support.

rsta2, 2018-04-08:
This is the same firmware I have tested successfully with Circle without U-Boot. So it seems, it is an U-Boot problem.

Can you please run the attached kernel image on the RPi 3B+ and use U-Boot to load it. I do not have a recent U-Boot installed. There are two different interfaces for the SD card (EMMC and SDHOST) and maybe U-Boot switches the interfaces from the firmware default setting, so that Circle cannot access the EMMC interface any more. The program shows the GPIO settings for GPIO 28-53. It can help to know these settings, to figure out if U-Boot is doing this. The output can be redirected to the serial interface using *cmdline.txt*.

[kernel7.img.zip](https://github.com/rsta2/circle/files/1887362/kernel7.img.zip)

jsanchezv, 2018-04-08:
The [patch in U-Boot](https://patchwork.ozlabs.org/patch/887255/).

From your test program:

GPIO028, 032, 033, 044, 045, 046 == Input
GPIO 029, 030, 031, 047 == Output
GPIO034 to 039 == ALT3
GPIO040 to 043 & 048 to 053 = ALT0

rsta2, 2018-04-08:
I have already seen the U-Boot patch. I think, this is not the whole story. They only change the SD card clock source from EMMC to CORE. This does not explain, why it is still working, when using Circle without U-Boot. Circle still uses the EMMC interface and the EMMC clock and it is working here.

Thanks for testing. I think, this enlightens what's happening there. U-Boot changes the default firmware settings for GPIO 48-53 from ALT3 (EMMC) to ALT0 (SDHOST) and does not switch this back on exit. It uses the SDHOST interface for SD card access now as Linux is doing. That's why you can probably boot Linux using U-Boot. But Circle needs access to the EMMC interface to be able to access the SD card. Changing this would need a completely new SD card driver.

Given that this may be a U-Boot problem, because it does not restore the GPIO settings to the state when U-Boot is started, I think about doing this in Circle itself. Unfortunately these GPIOs are not documented in the "BCM2835 Peripherals" document and I have to be careful. I will come back on this soon.

rsta2, 2018-04-08:
I wonder if you can do another test with the attached kernel image? It is the *addon/SDCard/sample* with a slightly modified EMMC driver, which sets GPIO34-39 to INPUT and GPIO48-53 to ALT3 so that the EMMC interface can be accessed by the driver. You should run this **on the RPi 3B+ only**. 

Unfortunately the setting of the pullup/pulldown resistors cannot be read back by the hardware, so I do not know how to set them. That's why I left them untouched. Because of this, there is some change, that it does not work, but maybe we are lucky.

[kernel7.img.zip](https://github.com/rsta2/circle/files/1887541/kernel7.img.zip)

jsanchezv, 2018-04-08:
YES!, another bug bites the dust... :)
I can see the root directory contents and Hello File! (Line 1 to 5).

rsta2, 2018-04-08:
OK, great. Now we know, it was the expected problem. Thanks for testing!

Unfortunately the fix will only work safely on the RPi 3 and 3B+, because the other models have different GPIO assignments. I don't know, if the problem occurs on the RPi 1, 2, Zero, CM (?) too, when using U-Boot.

I think, I will prepare a fix for the RPi 3 and 3B+ in the repo first. I want to complete this in the next week.

jsanchezv, 2018-04-08:
I've tested with last u-boot compiled for PI-2 and all works OK. No problem with uSD (what I can see).

rsta2, 2018-04-09:
OK, thanks. Then I think, a solution for RPi 3 (B and B+) should be enough for the moment.

rsta2, 2018-04-09:
I have added the workaround for the RPi 3B and B+ to the *develop* branch.

jsanchezv, 2018-04-09:
The develop branch works OK with new u-boot on PI-2 & PI-3 and access correctly to uSD. The only one configuration I can't test is PI-2 v1.2 (the PI-3 without WiFi/Bluetooth) because I don't own these model.

rsta2, 2018-04-09:
If there is a problem when using the RPi 2 v1.2 with U-Boot and Circle maybe we will hear from it later. Thanks again for testing.

closed 2018-06-12 by rsta2