rsta2/circle #71: Installation process - Guidance Please

Abrams88, 2018-05-10:
Hi there, 

I am a newbie, playing with Raspberry pi, so I am not familiar with Building and Installation processes.
Can anyone please explain me with more details, how I install and run(test) Circle on RPI2? 
I have build the source code, and I have mount my SD card on a Linux machine. 
What I have to do then? 
I have to rename the kernel.img to kernel7.img and replace the old one located at /boot dir?
And then what? I insert the SD card  on RPI2? 
I need a serial cable and bootup from U-boot? 
Sorry for my ignorance

rsta2, 2018-05-11:
If you want to use a SD card with Raspbian installed on it, you should do the following:

1. Rename the file *kernel7.img* on the mounted *boot* partition to *kernel7.img.linux*. Please note, that Raspbian will not boot any more then, until you rename the file *kernel7.img.linux* back to *kernel7.img*.
2. Copy the file *kernel7.img* from a ready built Circle sample to the *boot* partition of the SD card.
3. Unmount the SD card from you PC, put it into the Raspberry Pi and start it.

The Circle sample should run then. Normally you do not (!) need a serial USB adapter or U-Boot. This is only an option to speed-up the development process.

Please be sure, that you do not destroy any file on the *boot* partition of your SD card, if you want to use it later to run Raspbian again.

If you have a free SD card with a FAT partition, you can also use it for Circle. Go to the boot/ subdirectory of the Circle project and do `make` there. This will download the firmware files, needed to boot Circle. Just put these files on the empty SD card together with the *kernel7.img* file from the built Circle sample, unmount it and put the SD card into your RPi 2. It should work then.

IliasAbrams, 2018-05-11:
Thank you sir for your reply. 
 I appreciate that. 
Many thanks!

closed 2018-05-23 by rsta2