rsta2/circle #75: Updating kernel.img remotely

GiovanniCmpaner, 2018-08-02:
Hi everyone!

Is there a way right now to update kernel.img remotely, without having to take out the SD Card from the Raspberry?

I tought of handling a socket together with file operations, but I'm asking before to see if it is already available.

Thanks!

MaPePeR, 2018-08-02:
I think there is a way to do it with serial communication. See https://github.com/rsta2/circle/pull/58

rsta2, 2018-08-02:
Another possibility is to add a TFTP file server for the SD card root directory to your application as it is described [here](https://github.com/rsta2/circle/blob/master/addon/tftpfileserver/sample/README).

GiovanniCmpaner, 2018-08-02:
Thanks both, @MaPePeR and @ rsta2, very helpful links!

closed 2018-08-02 by GiovanniCmpaner