rsta2/circle #77: Support for ADS7846-based touchscreens ?

stephanedamo, 2018-08-06:
Hello

I have an HDMI touch screen which works great with Circle for display purposes, but ignores the touch screen. I noticed it's because the driver expects an FT5406 chip while my screen has an ADS7846 controller, communicating though USB.

There are already Arduino and linux libraries supporting it : 
- Arduino code : https://github.com/watterott/Arduino-Libs/blob/master/ADS7846/ADS7846.cpp
- Linux code : https://github.com/torvalds/linux/blob/master/drivers/input/touchscreen/ads7846.c

I wanted to port the arduino code to Circle but I struggle understanding how to properly setup the USB controller to do the required read/write operations. In the sample, there are no "bulk" USB examples showing how to perform raw read/write operations (or I missed something...)

Could you provide some help with this ? (or tell me if you have planned to support it soon)

Regards

rsta2, 2018-08-07:
Unfortunately the ADS7846 is not supported yet and I'm not planning support for it at the moment. If the ADS7846 is using USB bulk transfers and you want to develop a driver by yourself, you should have a look at the different USB Ethernet device drivers in the *lib/usb* subdirectory (smsc951x.cpp, lan7800.cpp or usbcdcethernet.cpp). They are all using USB bulk transfers and are relatively small. The device set-up is in the Configure() method, bulk-out transfers in SendFrame() and bulk-in transfers in ReceiveFrame(). If you have specific questions, you can ask them here.

stephanedamo, 2018-08-09:
Thank you for your answer

rsta2, 2018-09-26:
I had a closer look now into the sources referenced in your initial comment. It turned out, that the ADS7846 has a SPI interface, not a USB-based interface. Access to SPI devices is shown in *sample/23-spisimple*.

stephanedamo, 2018-10-01:
You're right, but it's the screen I use, it provides access to the touch only through USB or HDMI but no direct access to SPI. The touch screen ribbon seems to go to an STM32 microcontroller which acts probably as a wrapper between the two interfaces (or maybe it emulates the ADS7846). It's completely plug and play on Windows. I don't have much knowledge on USB devices unfortunately...

stephanedamo, 2018-10-01:
I deduced my screen uses ADS7846 or at least emulates it by looking at the driver the manufacturer provided (intended for Raspbian). If you're interested, I uploaded the files here : 

http://sdamo.io/touchscreen/

in ./LCD-show/boot/ some of the config files refers to the ADS7846.

rsta2, 2018-10-02:
It's really difficult if you do not have the driver source code for this proprietary touchscreen interface. I had a look on the uploaded files, but I didn't find something useful for this purpose.

stephanedamo, 2018-10-05:
It works without any driver installation on Windows, it's perfectly plug and play. To behave like that it may be a HID-compliant device, i'll try your CUSBHIDDevice and let you know if it's working

rsta2, 2020-01-31:
I guess, this feature is not of interest any more and it is currently not planned to support ADS7846-based touchscreens. So the issue can be closed.

closed 2020-01-31 by rsta2