rsta2/circle #79: I2S/PWM audio callback while processing on multiple cores

stephanedamo, 2018-08-10:
Hello,

I've managed to create a synthesizer using the CPWMSoundBaseDevice and CI2SSoundBaseDevice, however to get more polyphony I'd like to use the four CPU cores of my raspberry PI 3 B+.

I've basically created this system : 
- The audio callback (GetChunk) gets automatically called by your library
- Some global volatile variables are set, telling what has to be processed by each core
- The 4-core Run() loop (my sound class now also inherits from CMultiCoreSupport) checks the state of those global variables, and start processing if it should. Then it adds the results to a common buffer.
- A while loop in the audio callback makes the main core wait for the others to finish before returning. (this is done with a global counter incremented each time a core finished its work)

The problem is  that modifying variables in one core has no effect on the other cores, despite being global variables. Especially inside the audio callback, since I've managed to make global variables visible to any core in your sample #26 (cpustress) by using the _volatile_ keyword. But it doesn't seems to work inside CPWMSoundBaseDevice or CI2SSoundBaseDevice interrupts. I've tried separating the audio class from the multicore class without success.

Can you tell me if combining audio callbacks and multi-core is _theorically_ possible ?
If yes could you give me some tips on how to proceed ?

Best regards

rsta2, 2018-08-10:
Yes, it's possible to combine the audio callbacks with multi-core processing of the sound data. I have done this in my project [MiniSynth Pi](https://github.com/rsta2/minisynth), which is a virtual analog synthesizer for the Raspberry Pi based on Circle. I'm using the 4 cores of the RPi 2/3 somewhat similar as you want to do it for your synthesizer to increase the number of polyphonic voices.

Perhaps you should have a look at the *MiniSynth Pi* project, how this is done there. The synchronisation of the different cores is implemented in the class [CVoiceManager](https://github.com/rsta2/minisynth/blob/master/src/voicemanager.h). There is a comment in this header file, how this is basically working.

Please note that a value which is written from one core to a memory location is immediately visible from another core at this memory location. This is ensured by the hardware (snooping). But incrementing a variable in memory is not an atomic operation, even if the variable is declared *volatile*. It requires the steps "reading the current value from memory", "incrementing the value inside a register of a core" and "writing the new value to memory". If these steps are processed on one core in a row, another core, which is doing the same, may disturb the first core. It may require additional synchronisation (e.g. using a spin-lock) to prevent race conditions in this case. But because spin-locks are relatively slow, I only used simple memory reads and writes in the class *CVoiceManager*.

stephanedamo, 2018-08-10:
Thank you for this very detailed answer, I can't try your synth for now (missing usb keyboard) but I found a strange behavior. Using the code from your synth, I've created this very minimal example : 

[main.zip](https://github.com/rsta2/circle/files/2279514/main.zip)

It's a simple Test class running on several cores, line 46 there is a reboot() function which is never reached because the loop before never ends. It *should* end since the Run() function sets the status for each core to CoreStatusIdle, which breaks the loop condition. Do you see something obviously wrong in this example ? If it seems correct maybe you could try running it on your raspberry to see if we get different behaviors ?

rsta2, 2018-08-11:
I did some changes to your example, so that it is working now. Please see the comments in the attached file.

[main.zip](https://github.com/rsta2/circle/files/2280255/main.zip)

stephanedamo, 2018-08-13:
Very nice ! I was pretty sure it was a problem like initialization order or a missing class -- thank you for helping with this, it would be very hard to guess, especially those classes that need to be instanciated even if not directly used by the code.

rsta2, 2018-08-14:
You are welcome. Please note that you also have to create an instance of the *CTimer* class in case you want to use the USB driver. I left this out in the example above.

stephanedamo, 2018-08-20:
I have a quite interesting behavior. Same sound generation function running on all cores, but some of them have a loud white noise in background, depending on which core handle the voice. The noise gets lower if I lower the stress on the memory bus. It seems that setting a global volatile value (m_CoreStatus) and directly reading it creates some strange behavior. If I put something as simple as a unused variable between the write and the read, the noise level is lowered :

```
while (1)
{
	m_CoreStatus[nCore] = CoreStatusIdle;
	
	volatile unsigned dummy = 1;
	
	while (m_CoreStatus[nCore] == CoreStatusIdle)
	{
	}

	processVoices (nFirstVoice, nLastVoice, nCore);
}
```

If the dummy variable is put a before the "=", inside the while or after, it has no effect. When put in the right place, it has the same effect as the dummy loop you put in your example, saying some core wouldn't start if it wasn't there.

I guess I need to find the other places a dummy variable is needed to completely shut down this noise.

Do you have ideas on what is happening here ?

jsanchezv, 2018-08-20:
I can suggest to add a call to DataMemBarrier() (from synchronize.h) after every m_CoreStatus[nCore] modification.

stephanedamo, 2018-08-20:
It's working with DataMemBarrier() !

http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.faqs/ka14041.html

The reasons behind are more complicated than I thought. Thank you a lot for your help

rsta2, 2018-09-26:
I think, with the help by @jsanchezv your problem has been solved. So this can be closed.

closed 2018-09-26 by rsta2