rsta2/circle #80: Gpio FIQ Interrupt support

Matheus-Garbelini, 2018-08-15:
Hello @rsta2 I was wondering how to use gpio interrupt with FIQ. I've managed to use timer interrupt with FIQ because there's a attachFIQ function to it, but that's not true with gpio.

There's someting I could use for gpio FIQ interrupt without messing with other interrupts that are already using the FIQ manager?

rsta2, 2018-08-16:
There is a special class [CGPIOPinFIQ](https://github.com/rsta2/circle/blob/master/include/circle/gpiopinfiq.h), which has to be used, if you want to trigger a FIQ from a GPIO pin. Have a look at the [sample/30-gpiofiq](https://github.com/rsta2/circle/tree/master/sample/30-gpiofiq) for info on how to use it.

Please note that there can be only one FIQ interrupt source in the system at a time, which is caused by the hardware. There is another restriction so that *CGPIOPinFIQ* cannot be used, when *CGPIOManager* is active in Circle. That means that GPIO interrupts can only be handled using the IRQ (multiple GPIO pins allowed) or the FIQ (only one GPIO pin allowed), not using IRQ and FIQ at the same time.

Matheus-Garbelini, 2018-08-16:
Could I use multiple gpio fiq interrupts if I connect them to the same fiq function?
If so, I'd have to manage in te function how to detect from what gpio the interrupt was triggered?

rsta2, 2018-08-16:
No, this is not possible with *CGPIOPinFIQ*, which only works with one single GPIO pin. If you may use the IRQ this would work with *CGPIOPin* and *CGPIOManager*. Are you sure, you need the FIQ? If so, this would require a specific custom class.

Matheus-Garbelini, 2018-08-16:
@rsta2 I'm using 4 interrupts with irq right now, but I'd need to use fiq for those gpio to reduce the latency as minimum as possible. I'll look into creating a new class for it then. Thank you.

rsta2, 2018-08-17:
OK. Let me know, if you need help on that.

rsta2, 2018-09-26:
The initial question has been answered and I guess, you have already found a solution. So this can be closed.

closed 2018-09-26 by rsta2