rsta2/circle #81: Not an issue with your code but...

xlar54, 2018-08-20:
For others out there, I build with Cygwin on Windows.  Works fine, except that in the current version I have (arm-none-eabi-g++ 5.4.1), there appears to be a compiler bug (which I dont have the current text of, sorry).  The fix is to go into the lib/net directory and open up httpclient.cpp.  Add this pragma before the method name:

#pragma GCC optimize("O0")
THTTPStatus CHTTPClient::Request (THTTPRequestMethod  Method,
				  const char	     *pPath,
				  u8		     *pBuffer,
				  unsigned	     *pLength,
				  const char	     *pFormData)

rsta2, 2018-08-22:
Thanks for this info. I modified the title, so the entry can be found easier.

rsta2, 2018-09-26:
I will close this issue, because it's not an open bug in Circle. The issue can be found in the list of closed issues.

closed 2018-09-26 by rsta2