rsta2/circle #82: Fails to build on macOS High Sierra

cmarrin, 2018-09-12:
Rules.mk has a check for the version of make, requiring it to be at least 4.0. macOS High Sierra has 'GNU Make 3.81', so it fails to build. Commenting the test out makes everything build correctly. I'm using the arm-none-eabi tools installed from brew. I've also tried the package downloaded from the ARM site (which appears to be slightly newer) and it works as well. 

Are there systems where a 3.x version of make fails to build? Is the solution to allow 3.0 or above, or to add a test for some minimum version of macOS?

cmarrin, 2018-09-12:
Let me amend that. There was one other thing I needed to do. I changed STDLIB_SUPPORT from 1 to 0. Otherwise the link failed with uidiv errors. I assume this doesn't fail on other platforms. So I'm not sure why macOS would have anything to do with this. Seems like this is all down in the cross-platform libraries so it should be the same everything.

rsta2, 2018-09-13:
GNU make versions before 4.0 do not support the `!=` operator, which is used inside *Rules.mk* in case *STDLIB_SUPPORT* is not equal 0. See issue #67 .

I will modify *Rules.mk* so that this version check is not applied any more, if *STDLIB_SUPPORT* is equal 0.

rsta2, 2018-09-13:
I have done the described modification. It's on the *develop* branch and will be on *master* when the next release is ready. Thanks for reporting this issue.

cmarrin, 2018-09-13:
Thanks a lot. I'll check it out.

rsta2, 2018-10-21:
The modification is on *master* branch now.

closed 2018-10-21 by rsta2