rsta2/circle #87: Serial.Read() randomly crashing

stephanedamo, 2018-11-04:
Hello,

I've noticed after a lot of investigations that calling Serial.Read() may crash the Raspberry PI 3B+ (happens anywhere between 1min / 2 hours in my program).
It's random, but a way of making it crash very fast is to send serial data to the raspberry and call Read() in a loop without the program doing other things, like in this minimal example : 
[main.zip](https://github.com/rsta2/circle/files/2546028/main.zip)
I test it using MIDI data (sent by my computer through an USB/MIDI adapter, then fed into an optocoupler and some resistors then to the RPI's GPIO). I use this file because it has a lot of events so it also helps crashing it faster : 
[Goden_Axe_-_Stage_1.zip](https://github.com/rsta2/circle/files/2546041/Goden_Axe_-_Stage_1.zip)

**If no serial data is sent, Read() is fine and will never crash. The problems shows only when there is a least one byte to read.**

If you uncomment line 45 to make Read() calls less frequent, it will take more time for it to crash.

It's not dependant of the total bytes received, in my real program it crashed anywhere between 300KB/3MB of data received.

When it crashes, either the Raspeberry freezes, or the error is catch by your ExceptionHandler and it's always a Prefetch Abort error, FSR = 0xD (permission error according to the ARM docs)

I've tried your example #29 (miniorgan) and managed to crash it (I just rewrote the GetChunk with a loop that write zero's so it's not much busy and can crash in less than one minute). I don't get much Prefetch Abort errors here but freezes and sometimes this message just before : 

> interrupt.cpp(122): assertion failed: FIQData.pHandler == 0

Do you know what may happen ?
I can do more tests if you want, since it may not be that easy for you to setup.

stephanedamo, 2018-11-04:
Happen only when Serial uses FIQ

rsta2, 2018-11-05:
I can confirm, that there is a problem and this one is probably hard to find. Your slightly modified (serial baud rate changed to 38400 Bps, `m_Interrupt.Initialize()` must be called before `m_Timer.Initialize()`) *main.cpp* test program crashed here with a failed assertion in synchronize.cpp(103) after about 13 minutes continued sending data using a serial USB adapter and minicom on a PC. Without the second modification the system timer will not work, but this should not make a difference here. I used the `-Ofast` option, because I guess, you did this too (?). I have to do additional tests to try to find the cause of this issue. I will keep you informed.

stephanedamo, 2018-11-05:
I tried both with O1/O2/O3(Ofast) and the behavior is the same. I found a code that uses DMB/DSB instructions in the FIQ handler, I tried that too and made no difference.
It's a bit annoying if it takes 13 minutes to crash on your test system, have you tried sending more data through serial ? Mine crash in 0-2 seconds when a 2KB/s stream is sent, when the waiting loop of the example is commented.
I'll correct the timer thing for the next tests. Thanks !

rsta2, 2018-11-05:
Yes, the same problem arose here with `-O2`. I did some more tests and maybe I found something. Can you please replace the file *lib/synchronize.cpp* with the attached file and do some tests then (be sure to have the latest Circle version from the master branch). With the modification contained in this file, your test program was running for one hour here without problems, while receiving 14 MByte data.

[synchronize.cpp.zip](https://github.com/rsta2/circle/files/2548402/synchronize.cpp.zip)

stephanedamo, 2018-11-05:
At first I thought it worked but I was just lucky, I did multiple tests and it behaves the same.

[image]

I've just added Screen.Rotor() on this test, so I can see if it freezes, but it didn't, I only got Prefetch Aborts. PC and FAR are random, but SP/LR/PSR always have the same value.

I've tested with/without your new synchronize.cpp, and with/without SAVE_VFP_REGS_ON_FIQ, on the latest Circle version from master branch.

stephanedamo, 2018-11-05:
Found something.

It only crashes when ARM_ALLOW_MULTI_CORE is enabled. Not matter which synchronize.cpp I use.

If I disable ARM_ALLOW_MULTI_CORE then I have no crashes ( -- at least in a 5 min period, I can't say if it's completely safe or just more stable --)

Tested on Circle 36 with default compiler settings

rsta2, 2018-11-06:
There is another file attached, which has to replace *lib/spinlock.cpp*. Please use both new files for further tests. Without this new spin lock implementation it crashed here immediately, if ARM_ALLOW_MULTI_CORE has been enabled. I'm still running a test and had I had no more crashes after changing the spin lock now.

[spinlock.cpp.zip](https://github.com/rsta2/circle/files/2551915/spinlock.cpp.zip)

rsta2, 2018-11-06:
The last test with ARM_ALLOW_MULTI_CORE enabled was running successfully for one hour.

stephanedamo, 2018-11-06:
Hello,
With the new spinlock.cpp it doesn't crash anymore.
However the serial stops working after 0-5 seconds of use, you can check by printing a character on screen each time something is received with this code : 

```
while(1)
{
	u8 Buffer[20];
	int nResult = m_Serial.Read (Buffer, sizeof Buffer);
		
	if(nResult>0)
		m_Screen.Write("o", 1);
}
```

Strange fact : If I also print something when no serial data was received, it doesn't crash : 

```
while(1)
{
	u8 Buffer[20];
	int nResult = m_Serial.Read (Buffer, sizeof Buffer);
		
	if(nResult>0)
		m_Screen.Write("o", 1);
	else
		m_Screen.Write("_", 1);
}
```

Maybe there is some race condition or unfinished memory write somewhere, so making the CPU busy between the Read()'s prevents any problem ?

rsta2, 2018-11-06:
This problem seems to be sensitive to timing. With the second variant the serial hangs  too here after a longer while. I can see that the FIQ is disabled in that moment. That's why the serial interrupt handler cannot be called any more and receiving data stops. Currently I have no idea, what is the reason for this problem. Obviously there is a fundamental problem with the FIQ. FIQ support has been integrated into Circle relatively late and has not been tested extensively. Sorting this out may take longer. Perhaps you should use the IRQ for the serial interface for the moment, if this is possible.

rsta2, 2018-11-07:
This might give us more time, that's good. I also think the problem is somewhere in the spin lock implementation, but not sure. When I'm replacing this section in CSpinLock::Acquire() and Release() directly with EnterCritical() and LeaveCritical() the problem did not occur any more so far:

```
void CSpinLock::Acquire (void)
{
	if (m_nTargetLevel >= IRQ_LEVEL)
	{
		EnterCritical (m_nTargetLevel);
	}

	if (s_bEnabled)
	{
```

and

```
void CSpinLock::Release (void)
{
	...

	if (m_nTargetLevel >= IRQ_LEVEL)
	{
		LeaveCritical ();
	};
}
```

But I cannot explain why this happens. With ARM_ALLOW_MULTI_CORE enabled EnterCritical() is not used by itself in the test program, also not in the libraries anywhere. Perhaps I could update the spin lock implementation as mentioned above, if it passes further tests, but I want to know what went wrong here.

My E-mail address is noted in most of the source code files of Circle. You can write to it.

rsta2, 2018-11-07:
I have found a reason for the problem. The first part of CSpinLock::Acquire() looks like this:

```
void CSpinLock::Acquire (void)
{
	if (m_nTargetLevel >= IRQ_LEVEL)
	{
		asm volatile
		(
			"mrs %0, cpsr\n"
			"cpsid i\n"

			: "=r" (m_nCPSR[CMultiCoreSupport::ThisCore ()])
		);

		if (m_nTargetLevel == FIQ_LEVEL)
		{
			DisableFIQs ();
		}
	}
```

But it has to look like this:

```
void CSpinLock::Acquire (void)
{
	if (m_nTargetLevel >= IRQ_LEVEL)
	{
		u32 nCPSR;

		asm volatile
		(
			"mrs %0, cpsr\n"
			"cpsid i\n"

			: "=r" (nCPSR)
		);

		if (m_nTargetLevel == FIQ_LEVEL)
		{
			DisableFIQs ();
		}

		m_nCPSR[CMultiCoreSupport::ThisCore ()] = nCPSR;
	}
```

With the first implementation m_nCPSR[core] may be set by a first entry into the function. Then, before the FIQ is disabled, a FIQ occurs and the FIQ handler calls Acquire() too. m_nCPSR[core] is overwritten then and when the FIQ handler returned, the call to Release() in CSerialDevice::Read() works with the wrong m_nCPSR[core] value. m_nCPSR[core] can only be written, when the FIQ is already disabled.

There is another optimisation, when m_nCPSR is written, when the spin lock is already acquired. Then there needs to be only one variable, not an array, and the number of the CPU core needs not to be requested.

stephanedamo, 2018-11-07:
Nice ! Multithreaded problems are so hard to deal with. Well done ;)

rsta2, 2018-11-07:
Thanks! I'm now trying to find out, why the LeaveCritical() did not work stable without ARM_ALLOW_MULTI_CORE.

rsta2, 2018-11-08:
I think, I have found the reason for the single-core issue too. It was a similar problem in EnterCritical() and LeaveCritical(). I suppose, I have fixed both problems and tested it with your test program with ARM_ALLOW_MULTI_CORE disabled and enabled for one hour each without problems. I did also check that the FIQ from the serial device was working in this period. The updated sources are on the *develop* branch.

Thank you for being that stubborn to find the cause for this problem! That helped very much.

stephanedamo, 2018-11-08:
You're welcome. As a C programmer learning bare metal ARM and assembly, your library is awesome. I'm happy to help you making it even better !

The fact that the real program only crashed after hours of normal operation was a big part of the problem. I let it run during nights, checking on the mornings if it had survived, commenting some parts of the code and repeat until it didn't crash anymore, so I could get more useful informations before posting here.

rsta2, 2018-11-08:
I can imagine, which effort it was, to get this information. But without this information it would have been nearly impossible to sort this out. Unfortunately there isn't a comfortable solution to debug multi-core applications running with Circle. Circle needs users like you, who help to make it better. It would be very difficult to do this all alone. So thanks again.

rsta2, 2018-11-14:
I have prepared a hotfix release (36.1) for this issue.

closed 2018-11-14 by rsta2