rsta2/circle #88: Unable to change frame buffer resolution

jhhoward, 2018-11-14:
I am developing an x86 emulator using Circle and need to be able to change resolution at runtime (e.g. from 640x400 to 320x200)

From reading the documentation of the mailbox system, it looks like it should just be a case of setting the right tags. I attempted this by deleting my instance of CBcmFrameBuffer and recreating it with the updated resolution. However this causes a strange problem where the display flickers (almost looks like it is flickering between the two resolutions) and appears corrupted.

I tried explicitly releasing the frame buffer allocation using the code 0x48001 in CBcmFrameBuffer's destructor but this did not help. I also tried creating a Resize() function, which updates the InitTags and calls Initialize() but this has the same problem as deleting and recreating the CBcmFrameBuffer object.

rsta2, 2018-11-15:
Unfortunately I cannot reproduce the problem. The attached test program switches the frame buffer resolution continuously between 640x480 and 320x200 and waits 4 seconds after each switch. It deletes the instance of *CBcmFrameBuffer*, creates a new one and calls *Initialize()* on it. The program does work successfully on a RPi 3B+ with the latest firmware. Can you please have a look at this program. Perhaps you can find, what went wrong. Or you may modify it, so that it shows the problem.

[fbswitchtest.zip](https://github.com/rsta2/circle/files/2584633/fbswitchtest.zip)

jhhoward, 2018-11-15:
Thank you for the test program, it helped me track down what the actual problem was: it turned out that the code that was writing to the newly allocated frame buffer was writing outside of the allocated bounds which caused the corruption problem.

jhhoward, 2018-11-25:
I am reopening this issue as although framebuffer resizing is working correctly on the Raspberry Pi 3, the system appears to freeze when changing resolution on the Raspberry Pi Zero. I can reproduce the problem using the fbswitchtest program that you attached in your previous comment.

rsta2, 2018-11-26:
I will check this.

rsta2, 2018-11-26:
There was obviously a different behavior between RPi 3 and Zero. The problem was probably caused by some stale values in the static frame buffer init structure, when it was reused. Now this structure will be copied on each re-init to ensure, it is clean. It is working here on the RPi 3B and RPi Zero with this update. Please use the code from the branch *develop* for the moment. There will be a new release in a few weeks. The update will be on *master* then.

rsta2, 2018-12-01:
The update is on the master branch now.

jhhoward, 2018-12-01:
Thanks! Can confirm fixed.

closed 2018-12-01 by jhhoward